openrocket/openrocket #2108: In 3D view, clicking in blank area should not close config editor

neilweinstock, 2023-03-09:
This is a regression from 15.03.

When a component is selected, it is outlined in red in the design view.  Clicking outside the rocket in the blank area will cause the component to be unselected and the outline removed.

If the config editor is open at the same time, expected behavior is that it should stay open after the unselect operation.  This is the behavior in 15.03.

In 22.02 it works correctly in the side view, but in 3D view the unselect causes the config editor to close.  It should stay open.

SiboVG, 2023-03-09:
I'm a bit confused as to why you think the config dialog should remain open when no components are selected? To me, the current behavior is correct.

neilweinstock, 2023-03-09:
It is frequently beneficial to get rid of the red outlines when doing adjustments to a component.  The only way to do that is to unselect.  Janky perhaps, but it has always worked this way (and still does in the side view) and it is very useful.

neilweinstock, 2023-03-20:
Busy day!

I'll try to test this and other stuff tonight.

PR #2128 (2023-03-24): [#2108] Don't dispose config dialog when no components selected in 3D view
This PR fixes #2108.
closed 2023-03-24 by SiboVG