openrocket/openrocket #2109: This Fin marking guide is quite ugly...

KTesh, 2023-03-09:
Ok...  I haven't printed it, but it's pretty ugly if you ask me...

[Estes Red Nova (7266) Fin guide openrocket-10422572677963216927.pdf](https://github.com/openrocket/openrocket/files/10933190/Estes.Red.Nova.7266.Fin.guide.openrocket-10422572677963216927.pdf)

What I really want is a template for pasting onto foamcore or cardboard to keep the fins straight while they're being glued up...  You know...  Something like what you get from Payloadbay's tool section:

https://www.payloadbay.com/index.php?page=Tools&action=FINGUIDES

KTesh, 2023-03-09:
And...  Why can't I post a .ork file here?

JoePfeiffer, 2023-03-09:
I'm not having any luck trying to open the .png file to look at it.
[Estes Red Nova (7266)(2 Stage E to E Mod).zip](https://github.com/openrocket/openrocket/files/10934818/Estes.Red.Nova.7266.2.Stage.E.to.E.Mod.zip)

hcraigmiller, 2023-03-09:
The most significant problem with this design is that phantom and "trick" components are being used for **visual effect**, **_not for construction_**; overlapping launch lugs has been fixed with launch lug instances. When such components are place in-line on the same body tube, the text overlaps.

The secondary problem is that the user is including "comment" type information in the part name, making the part name excessively long. The user should relocate such information from the part name and to the part comment. 

Correcting for the foregoing, the revised fin marking guide is much less "ugly".

[image]

Reduced to essentials, this a duplicate issue and should be closed.

JoePfeiffer, 2023-03-09:
Duplicate of #41

closed 2023-03-09 by JoePfeiffer

KTesh, 2023-03-09:
Thanks for breaking it down for me...  Moving the info to the comments...  OUCH...  I've got a lot of files... My head hurts just thinking about that.

[Modified Rev_01]<https://na01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F68821492%2F224119518-5f3a8a47-79c9-4fc7-821e-7a3bd2a1f0fb.png&data=05%7C01%7C%7C7348841c191f4104812b08db20cbaca5%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C638139831362456835%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=IfSWDsU6qHe9jxvVGGOYuxblIbiIKAurX1Ha8Rcdi%2FM%3D&reserved=0>