openrocket/openrocket #2131: [Bug] Saving as a .tmp file

Voyhager3, 2023-03-21:
**Describe the bug**
A clear and concise description of what the bug is.
When saving a design file (as of yesterday), the file will no longer save as a .ork, only as a .tmp, the file still works, but it doesn't save the name, so finding the file to rename it is a problem. 

**To Reproduce**
Steps to reproduce the behavior:
No one else seems to be having this issue, but I am using a flash-drive (but also have been for the past year+) with plenty of storage. But it occurs no matter how I try to save it.

**Expected behavior**
A clear and concise description of what you expected to happen.
The file to save as a normal .ork file, not whatever 73047023984792374274.tmp is

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]
[image]

**ORK File**
If the problem is triggered by a particular ORK file, please zip it and attach it to this issue.

**Platform**
 - OS: [e.g. Windows10] windows 10
 - OpenRocket version (e.g. 22-02) (22-02)
 - Graphics card (if a graphics or display-related problem)

**Additional context**
Add any other context about the problem here.

hcraigmiller, 2023-03-21:
Could not reproduce.

OR Build: 2022-02
Microsoft Windows 11 Pro; 10.0.22621 Build 22621.1105; Windows Feature Experience Pack 1000.22638.1000.0
Java(TM) SE Runtime Environment 18.9 (build 11.0.18+9-LTS-195)

Voyhager3, 2023-03-21:
I figured out the error was caused by the save file name including illegal characters, sorry, my bad.

SiboVG, 2023-03-21:
> I figured out the error was caused by the save file name including illegal characters, sorry, my bad.

Actually this is still an interesting bug. I'm gonna re-open this and change the title. @Voyhager3 could you give us the illegal character(s) that caused the problem?

Voyhager3, 2023-03-21:
The illegal character I was using was the colon ":", but a list of other illegal characters I found were as follows
/ \ ? * " : < > | 
I don't know if its a global computer thing, or just an .ork thing.

PR #2135 (2023-03-24): [#2131] Throw warning message when save filename contains illegal characters
This PR fixes #2131 and opens a warning message when you try to save a file with illegal characters (/ \ ? * " : < > |). This is valid for any file saving action in OR (design file save dialog, sim export dialog...).

https://user-images.githubusercontent.com/11031519/227068556-46c3ad80-4b97-4923-be32-8a8225f86e3e.mov
closed 2023-03-24 by SiboVG