openrocket/openrocket #2136: Write unit tests for RASAero importing

SiboVG, 2023-03-23:
#2123 already wrote some basic unit tests for RASAero file importing, but this should be elaborated to:

- [ ] Test every parameter import of every RASAero rocket component (e.g. test whether the length of the body tube is correct or not). You can use the complex rocket design already imported in the unit tests in #2123.
- [ ] Test simulation importing (including launch site settings importing)
- [ ] Motor importing
- [ ] Recovery device importing (test all the special cases, see #2123)

musaaero, 2026-10-04:
"Hi @SiboVG, I would like to work on this issue and add unit tests for the RASAero importer."

SiboVG, 2026-10-04:
> "Hi @SiboVG, I would like to work on this issue and add unit tests for the RASAero importer."
> 

Sure, go ahead!