openrocket/openrocket #2137: [Bug] Import RASAero II needs to set overall mass and CG at each stage when simulations are present

hcraigmiller, 2023-03-23:
It looks like mass and CG override rules are needed.

NOTES:

- To create a flight simulation in RASAero II, the CG and loaded weight for each powered vehicle (sustainer, sustainer plus first booster together, and sustainer plus first and second boosters together) must be manually entered.
- RASAero II assumes that the user accurately measured the and correctly entered the loaded weight and CG values for each powered vehicle in a simulation.
- To avoid potential RASAero II value discrepancies between flight simulations (where the design has more than one flight simulation), OpenRocket should compute the mass and CG override values from the first (or most complete) RASAero II simulation.

O = OpenRocket
R = RASAero II
Mo = Mass override (including subcomponents)
M = Mass
S = Stage
S1 = Stage 1
E = Engine (motor)

Example: OMoS1 = RMS1 - OMS1E means set the OpenRocket Stage 1 mass including subcomponents to the _RASAero II Sustainer loaded weight_ minus the _OpenRocket mass of the Stage 1 motor_.

RULES:

**MASS OVERRIDES**:

1. For the first stage, override the OpenRocket Stage 1 mass including subcomponents with the _RASAero II Sustainer loaded weight_ minus the _OpenRocket mass of the Stage 1 motor_ (with only Stage 1 active, **the OpenRocket mass with motors should match the RASAero II sustainer total loaded weight**) FORMULA: OMoS1 = RMS1 - OMS1E; 
2. If the design has a second stage, override the OpenRocket Stage 2 mass including subcomponents with the _RASAero II Stage 2 loaded weight_ minus the _OpenRocket mass of the Stage 2 motor_  minus the _RASAero II Stage 2 loaded weight_ (with only Stage 1 and Stage 2 active, **the OpenRocket mass with motors should match the RASAero II first booster total loaded weight**). FORMULA: OMoS2 = (RMS2 - OMS2E) - RMS1
3. If the design has a third stage, override the OpenRocket Stage 3 mass including subcomponents with the RASAero II Stage 3 loaded weight minus the _OpenRocket mass of the Stage 3 motor_ minus the _RASAero II Stage 2 loaded weight_  (with all three stages active, **the OpenRocket mass with motors should match the RASAero II second booster total loaded weight**). FORMULA: OMoS3 = (RMS3 - OMS3E) - RMS2

RASAero II limits the number of stages to three.

**CG OVERRIDES**:

1. For the first stage, override the Stage 1 CG including subcomponents with the RASAero II CG for Stage 2 (with only Stage 1 active, **the OpenRocket Stage 1 CG should match the RASAero II sustainer CG**).
2. If the design has a second stage: to follow.
3. If the design has a third stage: to follow.

RASAero II limits the number of stages to three.

SiboVG, 2023-03-23:
We can't do that because RASAero has custom mass and CG **per simulation**, right? That's the reason why I couldn't implement it.

hcraigmiller, 2023-03-23:
> We can't do that because RASAero has custom mass and CG **per simulation**

That's why I suggested that the OpenRocket import reference ONLY the RASAero II _first_ simulation. Unless the user OpenRocket motor masses are wrong, the above formula should be very close when applied to any other simulations imported into OpenRocket. It's not ideal, but it's better than nothing, especially where a lot of the RASAero II files that I've looked at only have one simulation anyway.

PR #2173 (2023-04-04): [#2137] Import mass and CG from RASAero sims
This PR fixes #2137, and mass/CG information is now imported from the first RASAero simulation.
closed 2023-04-04 by SiboVG