openrocket/openrocket #2142: Body Tube and Inner Tube config editors are not consistent

neilweinstock, 2023-03-27:
[image]

On Body Tube, "length" appears on top (correct), while on Inner Tube, length appears at bottom (incorrect).

Also, is there a reason Inner Tube doesn't have a "filled" option for wall thickness?  It seems no more or less likely to need this than Body Tube, feels like they ought to be the same, no?

The one clear difference that seems correct is that Body Tube has an "automatic" option for diameter and Inner Tube does not, because it's not applicable.  everything else seems like it ought to be the same.

SiboVG, 2023-03-29:
> Also, is there a reason Inner Tube doesn't have a "filled" option for wall thickness? It seems no more or less likely to need this than Body Tube, feels like they ought to be the same, no?

No idea, I guess that's something to discuss with @JoePfeiffer and @hcraigmiller, but I would create a separate ticket for it.

neilweinstock, 2023-03-29:
Well, a filled coupler is the same as a bulkhead... in reality, inner tubes, centering rings, bulkheads, and couplers are all variations of the same thing i.e. tubes. 

A "filled" inner tube would probably be a dowel or a piece of threaded rod or something cylindrical used inside the rocket.

hcraigmiller, 2023-03-29:
It appears that the common criterion for other components with a "filled" option (currently only body components) is whether that component accepts subcomponents. Applying that approach, inner tubes should have a "filled" option but other inner components should not.

PR #2156 (2023-03-30): [#2142] Set length setting as first item
This PR fixes #2142.

[image]
closed 2023-03-30 by SiboVG