openrocket/openrocket #2168: [Bug] 3D Figure Fail, JAR 22.02 in Windows Portable

nvdaydreamer, 2023-04-01:
### Describe the bug

When attempting to view the 3D Figure of my model, it gave me a bug stack instead. So I tried it with the "simple" sample, and it did it again.

### To Reproduce

1. Run OpenRocket 22.02 JAR with OpenJDK Portable 17.0.6-10 from thumbdrive (Windows 10 host machine)
2. Open Stylis.ork, or any *.ork file. Edit, or don't.
3.  Select 3D Figure view ... and this happens.
...

---------- Exception stack trace ----------
java.lang.ArrayIndexOutOfBoundsException: Index -1 out of bounds for length 1 at jogamp.opengl.windows.wgl.awt.WindowsAWTWGLGraphicsConfigurationFactory.chooseGraphicsConfigurationImpl(WindowsAWTWGLGraphicsConfigurationFactory.java:171) at com.jogamp.nativewindow.GraphicsConfigurationFactory.chooseGraphicsConfiguration(GraphicsConfigurationFactory.java:424) at com.jogamp.opengl.awt.GLCanvas.chooseGraphicsConfiguration(GLCanvas.java:1513) at com.jogamp.opengl.awt.GLCanvas.addNotify(GLCanvas.java:609) at java.desktop/java.awt.Container.addNotify(Container.java:2804) at java.desktop/javax.swing.JComponent.addNotify(JComponent.java:4839) at java.desktop/java.awt.Container.addImpl(Container.java:1150) at java.desktop/java.awt.Container.add(Container.java:1001) at net.sf.openrocket.gui.scalefigure.RocketPanel.go3D(RocketPanel.java:279) at net.sf.openrocket.gui.scalefigure.RocketPanel.access$700(RocketPanel.java:92) at net.sf.openrocket.gui.scalefigure.RocketPanel$7.setSelectedItem(RocketPanel.java:337) at java.desktop/javax.swing.JComboBox.setSelectedItem(JComboBox.java:609) at java.desktop/javax.swing.JComboBox.setSelectedIndex(JComboBox.java:654) at java.desktop/javax.swing.plaf.basic.BasicComboPopup$Handler.mouseReleased(BasicComboPopup.java:946) at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:298) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389) at java.desktop/javax.swing.plaf.basic.BasicComboPopup$1.processMouseEvent(BasicComboPopup.java:551) at java.desktop/java.awt.Component.processEvent(Component.java:6391) at java.desktop/java.awt.Container.processEvent(Container.java:2266) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02
OpenRocket source: default
OpenRocket location: F:\PortableApps\OpenRocket\OpenRocket-22.02.jar
User-defined thrust curves location: C:\Users\portnew\AppData\Roaming\OpenRocket\ThrustCurves;F:\PortableApps\OpenRocket\MyThrustCurves
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
file.encoding=Cp1252
file.separator=\
java.class.path=F:\PortableApps\OpenRocket\OpenRocket-22.02.jar
java.class.version=61.0
java.home=F:\PortableApps\CommonFiles\OpenJDK
java.io.tmpdir=C:\Users\portnew\AppData\Local\Temp\
java.library.path=F:\PortableApps\CommonFiles\OpenJDK\bin;C:\WINDOWS\Sun\Java\bin;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\bin;C:\Perl\bin;C:\Python22\.;C:\Program Files\Microsoft SQL Server\130\Tools\Binn\;C:\Program Files\Java\jre7\bin;C:\Program Files\dotnet\;C:\Program Files (x86)\Pulse Secure\VC142.CRT\X64\;C:\Program Files (x86)\Pulse Secure\VC142.CRT\X86\;C:\Program Files (x86)\Common Files\Pulse Secure\TNC Client Plugin\;C:\Users\portnew\AppData\Local\Microsoft\WindowsApps;;C:\Users\portnew\AppData\Local\Programs\Microsoft VS Code\bin;.
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.6+10
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=Eclipse Adoptium
java.vendor.url=https://adoptium.net/
java.vendor.url.bug=https://github.com/adoptium/adoptium-support/issues
java.vendor.version=Temurin-17.0.6+10
java.version=17.0.6
java.version.date=2023-01-17
java.vm.info=mixed mode, emulated-client
java.vm.name=OpenJDK Client VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=Eclipse Adoptium
java.vm.version=17.0.6+10
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln15607012989481087962
line.separator=\u000d\u000a
native.encoding=Cp1252
os.arch=x86
os.name=Windows 10
os.version=10.0
path.separator=;
sun.arch.data.model=32
sun.awt.enableExtraMouseButtons=true
sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
sun.boot.library.path=F:\PortableApps\CommonFiles\OpenJDK\bin
sun.cpu.endian=little
sun.cpu.isalist=pentium_pro+mmx pentium_pro pentium+mmx pentium i486 i386 i86
sun.io.unicode.encoding=UnicodeLittle
sun.java.command=F:\PortableApps\OpenRocket\OpenRocket-22.02.jar
sun.java.launcher=SUN_STANDARD
sun.jnu.encoding=Cp1252
sun.management.compiler=HotSpot Client Compiler
[... 196 lines cut ...]

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[Stylis.zip](https://github.com/openrocket/openrocket/files/11130156/Stylis.zip)

### OpenRocket version

22.02 

### What platform are you running on?

Windows

### Additional context

Probably just some weirdness about an entirely portable environment. I won't miss 3D if fixing this is too much. 

Anyway, thanks and keep up the good work. See you 'round Facebook,
Bob Portnell
Sparks, NV

SiboVG, 2023-04-01:
This is a known issue, see #1364. You can either use Java 11, or run the JAR with `java --add-exports=java.base/java.lang=ALL-UNNAMED --add-exports=java.desktop/sun.awt=ALL-UNNAMED --add-exports=java.desktop/sun.java2d=ALL-UNNAMED -jar OpenRocket.jar`.

closed 2023-04-01 by SiboVG

nvdaydreamer, 2023-04-01:
Oh, sorry! My bad. Thanks for the workarounds.

SiboVG, 2023-04-01:
> Oh, sorry! My bad. Thanks for the workarounds. (The second one works just fine for me.)

No problem, have fun using OpenRocket!