openrocket/openrocket #2180: Uncaught exceptions while running flight simulations

DanCrank, 2023-04-06:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Loaded this .ork file, which had been repeatedly edited in the old (15.03?) version of OpenRocket but never before opened in the new OpenRocket: https://danno.org/RocketBlogFiles/sims/blackwidow.ork
2. Saw a warning about the Loki L1040 motor having a "different thrust curve than the original" (?) but ignored it. I do not currently have a custom thrust curve for this motor, but may have had one in the past (not sure). I usually select "save all data" when I save, so I guess all the thrust curves are saved in the ORK file...?
3. Went to the simulation tab, selected all simulations and hit the run button
4. Got the uncaught exception below
5. After capturing the exception, clicked OK and got a second exception. At this point, the dialog behind the exception said it was running simulation #8. I didn't capture the second exception but I assumed it was a second occurrence of the same.

If possible, please send us the .ork file that caused the bug. (linked above)

Include your email address (optional; it helps if we can contact you in case we need additional information): danno@danno.org

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120) at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:445) at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:109) at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847) at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857) at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317) at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233) at java.desktop/java.awt.Dialog.show(Dialog.java:1070) at java.desktop/java.awt.Component.show(Component.java:1716) at java.desktop/java.awt.Component.setVisible(Component.java:1663) at java.desktop/java.awt.Window.setVisible(Window.java:1031) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005) at net.sf.openrocket.gui.main.SimulationPanel.runSimulation(SimulationPanel.java:361) at net.sf.openrocket.gui.main.SimulationPanel.access$900(SimulationPanel.java:74) at net.sf.openrocket.gui.main.SimulationPanel$RunSimulationAction.actionPerformed(SimulationPanel.java:573) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:297) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) Caused by: java.lang.NullPointerException at net.sf.openrocket.models.atmosphere.InterpolatingAtmosphericModel.getConditions(InterpolatingAtmosphericModel.java:40) at net.sf.openrocket.simulation.AbstractSimulationStepper.modelAtmosphericConditions(AbstractSimulationStepper.java:36) at net.sf.openrocket.simulation.RK4SimulationStepper.calculateFlightConditions(RK4SimulationStepper.java:483) at net.sf.openrocket.simulation.RK4SimulationStepper.calculateForces(RK4SimulationStepper.java:421) at net.sf.openrocket.simulation.RK4SimulationStepper.calculateAcceleration(RK4SimulationStepper.java:324) at net.sf.openrocket.simulation.RK4SimulationStepper.computeParameters(RK4SimulationStepper.java:290) at net.sf.openrocket.simulation.RK4SimulationStepper.step(RK4SimulationStepper.java:121) at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulateLoop(BasicEventSimulationEngine.java:178) at net.sf.openrocket.simulation.BasicEventSimulationEngine.simulate(BasicEventSimulationEngine.java:119) at net.sf.openrocket.document.Simulation.simulate(Simulation.java:406) at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:55) at net.sf.openrocket.gui.simulation.SimulationWorker.doInBackground(SimulationWorker.java:24) at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304) at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264) at java.desktop/javax.swing.Swi [... cut ...]
87867 166.324 DEBUG (SimulationRunDialog.java:455) Setting progress to 100 (real 100)
87868 166.324 DEBUG (SimulationRunDialog.java:241) Progressbar value 82
87869 166.327 DEBUG (SimulationRunDialog.java:408) Simulation done
87870 166.327 DEBUG (SimulationRunDialog.java:455) Setting progress to 100 (real 100)
87871 166.327 DEBUG (SimulationRunDialog.java:241) Progressbar value 82
87872 170.992 USER (SwingExceptionHandler.java:220) User requested sending bug report
87873 170.995 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2023-04-06:
>     1. Loaded this .ork file, which had been repeatedly edited in the old (15.03?) version of OpenRocket but never before opened in the new OpenRocket: https://danno.org/RocketBlogFiles/sims/blackwidow.ork

Thank you for including the .ork file

> 2. Saw a warning about the Loki L1040 motor having a "different thrust curve than the original" (?) but ignored it. I do not currently have a custom thrust curve for this motor, but may have had one [...]

That's not a problem --- this turns up with some 15.03 designs, and isn't important.
 
Looking at the stack trace, it appears to be hitting a null pointer when calculating atmospheric pressure. Not sure why...  @SiboVG it looks like you've done some work on the code near where the exception is happening?

FWIW, I'm not having any luck duplicating the error.  Could you possibly run the sims one at a time to verify that the issue is really in #8?

DanCrank, 2023-04-06:
Thanks for the quick reply. I'm on a different machine than I was earlier, for what it's worth (both are Windows 11, though).

On this machine, I see the exception repeatably only when the entire list of simulations is run, and only the first time I run it. Now I am only seeing a single exception, while it is apparently running simulation 7. I don't get any exception when a single simulation is run (either first or subsequent runs) or when a smaller range of simulations is run (6 through 9 was what I tried).

Also, if I save the file, close and reopen OR and load the new file, I can run the entire list of sims without an exception. I made the OR 22.02-saved version available at https://danno.org/RocketBlogFiles/sims/blackwidow2.ork

This is in no way a blocking issue, and I'm fine with closing it as can't-reproduce. I just wanted to pass it along. :)

SiboVG, 2023-04-07:
> Looking at the stack trace, it appears to be hitting a null pointer when calculating atmospheric pressure. Not sure why... @SiboVG it looks like you've done some work on the code near where the exception is happening?

Well yeah, in 2021 😳, I don't recall it anymore... I just "fixed" it back then by re-computing the layers if one was null. I guess we could implement the same thing again. But I never found the root cause why this occurred.

JoePfeiffer, 2023-04-07:
I thought your q&d fix was still in the code?

Based on his update it sounds like a race condition, which...  ewwww.

hcraigmiller, 2023-04-07:
Unable to replicate in either 2022.02 or the current unstable.

On design open, shows warning about the Loki L1040 motor having a different thrust curve; ignored. Repeated closing OR and restarting, then opening design file, and re-running all simulations as a block. No error in either 2022.02 or current unstble after more than a dozen repeats.

Microsoft Windows 11 Pro; 10.0.22621 Build 22621.1105; Windows Feature Experience Pack 1000.22638.1000.0
Java(TM) SE Runtime Environment 18.9 (build 11.0.18+9-LTS-195)

SiboVG, 2023-04-07:
> I thought your q&d fix was still in the code?

It is, that's the recomputing of the layers, but it's only present for 2 calls in the if-conditions, not the `levels[n]` call.
[image]

So can just add a Q&D fix again, but for the `levels[n]` call.

FYI: I was able to reproduce this 2 times using OR 22.02 (error came straight away), but after those 2 times was unable to reproduce it.

closed 2023-04-12 by SiboVG