openrocket/openrocket #2187: Bug Report

TacoPabloTaco, 2023-04-08:
java.lang.NullPointerException at net.sf.openrocket.gui.scalefigure.FinPointFigure.updateSubjectDimensions(FinPointFigure.java:377) at net.sf.openrocket.gui.scalefigure.AbstractScaleFigure.updateFigure(AbstractScaleFigure.java:195) at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig.updateFields(FreeformFinSetConfig.java:429) at net.sf.openrocket.gui.configdialog.ComponentConfigDialog.componentChanged(ComponentConfigDialog.java:186) at net.sf.openrocket.rocketcomponent.Rocket.notifyAllListeners(Rocket.java:646) at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:552) at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:561) at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:2262) at net.sf.openrocket.rocketcomponent.RocketComponent.fireComponentChangeEvent(RocketComponent.java:2274) at net.sf.openrocket.rocketcomponent.FreeformFinSet.setPoint(FreeformFinSet.java:354) at net.sf.openrocket.gui.configdialog.FreeformFinSetConfig$FinPointScrollPane.mouseDragged(FreeformFinSetConfig.java:534) at java.desktop/java.awt.Component.processMouseMotionEvent(Component.java:6683) at java.desktop/javax.swing.JComponent.processMouseMotionEvent(JComponent.java:3360) at java.desktop/java.awt.Component.processEvent(Component.java:6404) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4564) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
User-defined thrust curves location: C:\Users\Acer\AppData\Roaming\OpenRocket\ThrustCurves
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\Acer\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files\Oculus\Support\oculus-runtime;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files\Intel\Intel(R) Management Engine Components\iCLS\;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;D:\MatLab\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\NVIDIA Corporation\NVIDIA NvDLISR;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;D:\nTop\;C:\Users\Acer\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.17+7-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.17
java.version.date=2022-10-18
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.17+7-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln4428945555757410886
line.separator=\u000d\u000a
os.arch=amd64
os.name=Windows 10
os.version=10.0
path.separator=;
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
sun.boot.library.path=c:\program files\openrocket\jre\bin
sun.cpu.endian=little
sun.cpu.isalist=amd64
sun.desktop=windows
sun.io.unicode.encoding=UnicodeLittle
sun.java.command=C:\Program Files\OpenRocket\OpenRocket.exe
sun.jnu.encoding=Cp1252
[... 281 lines cut ...]
9020 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
9021 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
9022 314.542 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
9023 314.542 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
9024 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
9025 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
9026 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
9027 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
9028 314.542 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
[... cut ...]
9040 314.542 WARN (DoubleModel.java:1006) DoubleModel[constant=0.3312581544539299] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@11cc78ca]
9041 314.542 WARN (DoubleModel.java:1006) DoubleModel[constant=0.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@11cc78ca]
9042 322.491 USER (SwingExceptionHandler.java:220) User requested sending bug report
9043 322.491 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2023-04-08:
Thank you for your report.  Please let us know the exact steps you followed that led to the exception, and if you had gotten to the point of saving your rocket before the editing session that caused the crash please upload the .ork file.  It'll be easiest to zip the file first, as github won't let us upload files with extensions other than a limited set that doesn't include .ork.

SiboVG, 2023-04-20:
I was able to reproduce this in 22.02:
1. Open a new design
2. Add a body tube
3. Add a freeform fin set to that body tube
4. Edit one of the points of the fin set
5. Click "Split fins" --> Exception

This bug has already been fixed (don't remember where, but I do remember working on this issue).

closed 2023-04-20 by SiboVG