openrocket/openrocket #2188: Overriding subcomponents makes for a very ugly component tree

neilweinstock, 2023-04-10:
I appreciate the thought process that went into this last release but IMHO this is really not appealing, as it makes the whole component tree harder to read, and just generally ugly:
[image]

I'm not sure how to fix it yet.  Two thoughts I had were 
1) make the grey subcomponent override icons *smaller*
2) Put the override icons in their own column at the left
3) Do something else on the left side.  I'll try to mock something up.

neilweinstock, 2023-04-10:
Some mockups:
[image]

I'll do more to show at least one other override configuration.

neilweinstock, 2023-04-10:
I suppose that technically, this is an option as well, but I don't think it's better than any of the others.
[image]

SiboVG, 2023-04-10:
It's not possible to put stuff in the left column + I'm also not a fan of it.

neilweinstock, 2023-04-10:
If so then maybe we can start off by just smallifying the icons, which would be better than nothing.

JoePfeiffer, 2023-04-10:
I honestly don't see the problem with the current component tree display. I don't find the current override icons to be ugly, and I don't see them as making the tree harder to read.  Even if it were possible to move the icons to their own column, moving them farther from the components they refer to would, I believe, make it harder to associate them with the components.  I don't have any real objection to making them smaller, just so they stay large enough to still be easily visible.

neilweinstock, 2023-04-10:
Current appearance really bugs me.  We'll try some smaller icons; I think they can be kept completely readable.  If folks don't agree we won't do it.

SiboVG, 2023-04-10:
> Current appearance really bugs me. We'll try some smaller icons; I think they can be kept completely readable. If folks don't agree we won't do it.

I think it's a good suggestion to make those icons smaller, I can experiment with it a bit to see what works.

neilweinstock, 2023-04-10:
If you can wait, I'll have a draft PR up by tonight.

SiboVG, 2023-04-10:
> If you can wait, I'll have a draft PR up by tonight.

Sure. Can you use the `Icons.getScaledIcon` when scaling down the icons, instead of uploading new images? So, in `swing/src/net/sf/openrocket/gui/main/componenttree/ComponentTreeRenderer.java`:
```Java
double overrideSubcomponentIconScale = 0.8;
if (c.getMassOverriddenBy() != null) {
	p.add(new JLabel(Icons.getScaledIcon(Icons.MASS_OVERRIDE_SUBCOMPONENT, overrideSubcomponentIconScale)));
} else if (c.isMassOverridden()) {
	p.add(new JLabel(Icons.MASS_OVERRIDE));
}
if (c.getCGOverriddenBy() != null) {
	p.add(new JLabel(Icons.getScaledIcon(Icons.CG_OVERRIDE_SUBCOMPONENT, overrideSubcomponentIconScale)));
} else if (c.isCGOverridden()) {
	p.add(new JLabel(Icons.CG_OVERRIDE));
}
if (c.getCDOverriddenBy() != null) {
	p.add(new JLabel(Icons.getScaledIcon(Icons.CD_OVERRIDE_SUBCOMPONENT, overrideSubcomponentIconScale)));
} else if (c.isCDOverridden()) {
	p.add(new JLabel(Icons.CD_OVERRIDE));
}
```

Makes it easier for you to experiment with changing the scale + ensures we won't have to do actual image editing.

neilweinstock, 2023-04-10:
I could, but my hand-edited icons will be sharper and nicer.  At small
sizes, scaling becomes very much sub-optimal, and if we're going to be
evaluating for readability then I'd hate to be handicapped by the scaling
algorithm.  And these are only three small monochrome images, very very
easy to modify.  For now, for demonstration/mockup purposes, I'll just
throw in my edited images and we can decide what to do next based on the
feedback.

neilweinstock, 2023-04-10:
Instead of a draft PR here are mockups for evaluation.
[image]

neilweinstock, 2023-04-11:
After staring for a while I like option 3 best.  It is still (IMHO) very clear, but noticeably de-emphasizes the overridden subcomponents.  But I would settle for (2) if I had to.

SiboVG, 2023-04-11:
I can go along with no. 3.

SiboVG, 2023-05-16:
@neilweinstock could you open a PR with the nr 3 icons?

neilweinstock, 2023-05-16:
Thank you for the reminder.

Do I need to do anything other than swap the new images in?

PR #2216 (2023-05-18): Downsized subcomponent override icons
This PR swaps in smaller subcomponent override icons to fix #2188.
closed 2023-05-18 by SiboVG