openrocket/openrocket #2200: [Feature Request] Support boattails

SiboVG, 2023-04-23:
Some users on the r/rocketry Discord have requested the support for a boattail component: a transition at the aft end of a stage that is **nested in the next stage**. The best way of supporting that now is to add an inline podset (no radial offset, and one pod instance) with a transition and then move that transition to the aft end of the stage. But that probably doesn't simulate all that well...

For reference, this is a boattail in RASAero:
[image]

JoePfeiffer, 2023-04-23:
That's going to be interesting, since a pretty deep assumption is that external components don't overlap....

Maybe have a notion of an occluded external component?  Has no aero effect when an adjacent stage is active, has an effect if not?

SiboVG, 2023-04-24:
> Maybe have a notion of an occluded external component? Has no aero effect when an adjacent stage is active, has an effect if not?

I was thinking something along those lines, yes.

hcraigmiller, 2023-04-25:
To be compatible with RASAero II, the booster into which a boattail is inserted must begin with a body tube (transitions are prohibited) having the same diameter as the prior body tube (forward diameter of the boattail).

JoePfeiffer, 2023-04-25:
We've got something like a zillion incompatibilities with RASAero, and I'm fine with one more. If someone creates a design that doesn't export, so be it.  No reason to limit what we can do because of it.