openrocket/openrocket #2203: error de simulacion

Diosleo, 2023-04-26:
Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)
al momento de realizar la simulación de vuelo se produce un error de sofware
1.simular el vuelo del cohete
2.colocar los motores con el que se va a trabajar

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
net.sf.openrocket.gui.main.SwingExceptionHandler$InternalException: An exception occurred during the simulation at net.sf.openrocket.gui.main.SwingExceptionHandler.handleErrorCondition(SwingExceptionHandler.java:120) at net.sf.openrocket.gui.simulation.SimulationRunDialog$InteractiveSimulationWorker.simulationInterrupted(SimulationRunDialog.java:445) at net.sf.openrocket.gui.simulation.SimulationWorker.done(SimulationWorker.java:109) at java.desktop/javax.swing.SwingWorker$5.run(SwingWorker.java:750) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.run(SwingWorker.java:847) at java.desktop/sun.swing.AccumulativeRunnable.run(AccumulativeRunnable.java:112) at java.desktop/javax.swing.SwingWorker$DoSubmitAccumulativeRunnable.actionPerformed(SwingWorker.java:857) at java.desktop/javax.swing.Timer.fireActionPerformed(Timer.java:317) at java.desktop/javax.swing.Timer$DoPostEvent.run(Timer.java:249) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117) at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:190) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:235) at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:233) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:233) at java.desktop/java.awt.Dialog.show(Dialog.java:1070) at java.desktop/java.awt.Component.show(Component.java:1716) at java.desktop/java.awt.Component.setVisible(Component.java:1663) at java.desktop/java.awt.Window.setVisible(Window.java:1031) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1005) at net.sf.openrocket.gui.main.SimulationPanel.plotSimulation(SimulationPanel.java:297) at net.sf.openrocket.gui.main.SimulationPanel.access$1100(SimulationPanel.java:74) at net.sf.openrocket.gui.main.SimulationPanel$PlotSimulationAction.actionPerformed(SimulationPanel.java:609) at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1967) at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308) at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405) at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262) at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279) at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635) at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342) at java.desktop/java.awt.Component.processEvent(Component.java:6400) at java.desktop/java.awt.Container.processEvent(Container.java:2263) at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918) at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547) at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90) Caused by: java.lang.NullPointerException

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
User-defined thrust curves location: C:\Users\user PC\AppData\Roaming\OpenRocket\ThrustCurves
JOGL version: 2.4.0-rc-20210111
Current default locale: es_ES
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\USERPC~1\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\Common Files\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft SQL Server\Client SDK\ODBC\110\Tools\Binn\;C:\Program Files (x86)\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Microsoft SQL Server\120\DTS\Binn\;C:\Program Files (x86)\Windows Kits\8.1\Windows Performance Toolkit\;C:\Users\user PC\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.17+7-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.17
[... cut ...]
38328 1460.407 DEBUG (SimulationRunDialog.java:241) Progressbar value 0
38329 1460.407 ERROR (SwingExceptionHandler.java:119) An exception occurred during the simulation
> java.lang.NullPointerException
38330 1460.407 INFO (SwingExceptionHandler.java:142) Running in EDT, showing dialog
38331 1460.407 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
38332 1550.456 USER (SwingExceptionHandler.java:220) User requested sending bug report
38333 1550.456 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2023-04-26:
Hi, thank you for your report, could you please send us the .ork that caused the issue?

Hola, gracias por su informe, ¿podría enviarnos el .ork que causa el problema?

Diosleo, 2023-04-28:
Enviado desde Correo para Windows De: Sibo Van GoolEnviado: miércoles, 26 de abril de 2023 13:43Para: openrocket/openrocketCC: Diosleo; AuthorAsunto: Re: [openrocket/openrocket] error de simulacion (Issue #2203) Hi, thank you for your report, could you please send us the .ork that caused the issue?Hola, gracias por su informe, ¿podría enviarnos el .ork que causa el problema?

SiboVG, 2023-04-28:
@Diosleo your message didn't really get through, could you try to upload the .ork file again? You can just zip it and upload it here.

Diosleo, 2023-04-29:
Enviado desde Correo para Windows De: Sibo Van GoolEnviado: viernes, 28 de abril de 2023 10:16Para: openrocket/openrocketCC: Diosleo; MentionAsunto: Re: [openrocket/openrocket] error de simulacion (Issue ***@***.*** your message didn't really get through, could you try to upload the .ork file again? You can just zip it and upload it here.

SiboVG, 2023-04-29:
Hi @Diosleo it still doesn't work. If sending the file over GitHub does not work, you can try sending it in an email to openrocket-bugs@lists.sourceforge.net. Thank you for your patience.

Diosleo, 2023-04-29:
https://drive.google.com/file/d/1R0EbKRrq1iLP0bNbF7It6RgIbMNu0wKR/view?usp=sharing

SiboVG, 2023-04-30:
I'll post the .ork file here, thanks @Diosleo!
[cohete 3.ork.zip](https://github.com/openrocket/openrocket/files/11361124/cohete.3.ork.zip)

SiboVG, 2023-05-06:
Okay, the issue is that your motor is too weak for the rocket, and OpenRocket throws an exception that your motor burnt out before your rocket even reached lift-off. Decreasing the rocket's mass or choosing a more powerful motor makes the error disappear.

But the main issue is that the burnout before lift-off exception should be shown to the user more clearly, as a simulation error instead of a BugException.

PR #2211 (2023-05-10): [#2203] Handle SimulationException separately
This PR fixes #2203. The problem was that the SimulationException in `computeCoastTime()` was handled as a general Exception, which caused the simulation error dialog to fail.

Result of this PR when running the sim:
[image]
closed 2023-05-10 by SiboVG