openrocket/openrocket #2213: [Feature Request] <ability to simulate a lander>

vyvizz, 2023-05-10:
We want to simulate a lander( model rocket) using a model rocket motor. It would be nice to have some intuition using openrocket to simulate backwards. Basically, the flight would start from the apogee and end at 0 altitudes with a propulsive landing.

JoePfeiffer, 2023-05-13:
This is pretty far outside our main development line.  There's some chance you might be able to implement it using the extension mechanism (see https://wiki.openrocket.info/Simulation_Extensions ) but I don't think the current development team is going to be able to get to it any time soon.

vyvizz, 2023-05-13:
I thought technically if you could start the simulation from altitude to simulate only drop that would be sufficient. That way you could put the rocket reverse to the original design and use a negative thrust curve.