openrocket/openrocket #2217: Major problem in tube fin set mass calculation

neilweinstock, 2023-05-23:
See attached ORK file:

- The body tube is 15" long.  Calculated mass is .827 oz.
- The tube fins are a set of 5, each 3" long.  Calculated mass is 4.3 oz.
- Both are based on the same library component (BMS T60-34).

There seems to be an extra multiplication by the number of fins (i.e. the mass of one fin is getting multiplied by 5 *twice*).  Pretty much confirmed this by experimenting with the number of fins in the set.

[tube_fins_mass.zip](https://github.com/openrocket/openrocket/files/11546443/tube_fins_mass.zip)

SiboVG, 2023-05-23:
FYI: this is only present in the latest unstable, not in 22.02.

neilweinstock, 2023-05-23:
OK, in that case it proves beyond doubt that at some point I swapped in an unstable build on my Windows machine.

Even if only in unstable, definitely needs fixing, but glad to hear it is not affecting 22.02 users.

JoePfeiffer, 2023-05-23:
Hmm, not sure when that would have snuck in... I'll look for that too

PR #2219 (2023-05-25): [#2217] Fix mass issues for fin sets (+ CM of launch lug and rail buttons)
This PR fixes #2217. The was caused by PR #2053 - I didn't realize that componentVolume was that of all the instances instead of a single instance.

Additionally, I noticed an issues with launch lugs and rial buttons where the y and z location of the CG wasn't right. If you created a new design with just a body tube and rail button/launch lug and overdid the mass of the body tube to 0, the CG should be located solely at the rail button/launch lug, which wasn't the case. This is now fixed.
closed 2023-05-25 by SiboVG