openrocket/openrocket #2221: Add units option for "Stability margin" in plot dialog

neilweinstock, 2023-05-23:
Right now "Stability Margin Calibers" is a Y-axis type in the sim plot dialog; it offers no units because the units are baked into the name.

Instead it should just be called "Stability Margin", with two unit options: calibers or % of length.

kjhambrick, 2025-03-15:
I wonder if length as ( CG - CP ) might be useful as well ?

One could calculate restoring Torque on the rocket without calculating the lever arm length from calibers or percentages.

-- kjh

luisroliveira, 2025-12-05:
Hello neilweinstock, good afternoon! Could you help me? I was able to add the units when selecting "Stability margin calibers," but I am having a problem when trying to change the variable's name in the messages.properties file, as if there is some dependency in the code on the variable's value. When I set it to <FlightDataType.TYPE_STABILITY = Stability margin>, "Stability margin" stops being an option to select. Do you know if the variable name in messages.properties could have any influence on this?

It looks like this when I leave the variable value as default:

[image]

This happens when I change the variable value:

[image]

luisroliveira, 2025-12-06:
All set! I just created a branch and pushed the changes. Here’s the link to the branch:
https://github.com/luisroliveira/openrocket/tree/feature_unitStabilityMargin

JoePfeiffer, 2025-12-10:
> To convert calibers to a percentage, is it enough to simply divide by 100, or is there an additional calculation required?

It's a different calculation -- instead of dividing by the rocket diameter, you divide by its length so it's
100 * (CP - CG) / length.

To the best of my recollection, it's different from all the other data types because the units are based on a characteristic of the rocket itself, and it ends up being a slightly different calculation as a result.

JoePfeiffer, 2025-12-11:
Oh, something else -- you should probably open a PR for your changes to keep the solution separate from the problem; mark it as "draft" until it's ready to go.