openrocket/openrocket #2222: ORK files opened via file association handled differently

neilweinstock, 2023-05-25:
On both Windows and Mac, files opened via file association (i.e.., double-clicking on the file on the desktop or whatever) are handled differently from those opened through File->Open.  There are two issues.

1) On both Windows and Mac, files opened via file association do not make it onto the "Open recent" submenu.
2) On Windows only, if you have an open instance of OR, then opening a file via association will open a completely new and separate OR instance, rather than simply opening the file in the existing open instance.

Although I am running an unstable build at the moment, I believe this all applies to the release version of 22.02.

SiboVG, 2023-06-20:
> 2\. On Windows only, if you have an open instance of OR, then opening a file via association will open a completely new and separate OR instance, rather than simply opening the file in the existing open instance.

This is already documented in #1737.

neilweinstock, 2023-06-20:
Well how am I supposed to keep track of what some internet rando filed 8 months ago?

(in other words:  oops ;))

PR #2230 (2023-06-27): [#2222] Fix file association Open Recent
This PR fixes #2222. Opening a file with the file association should now also cause that file to open in the recent design files.
closed 2023-06-27 by SiboVG