openrocket/openrocket #2234: [Feature Request] "Design Notes" text box addition

creilly27, 2023-06-26:
**Is your feature request related to a problem? Please describe.**
I often will come up with a plan to change my rocket design without the time to actually make the changes.  This leads me to create separate text documents with notes on changes I would like to implement in the future.  

**Describe the solution you'd like**
A popup text box window that appears when a button labeled something along the lines of "Design Notes" would be very helpful.  The comments sections are already useful, but they assume that you remember to check the comments on a particular component.

Thank you so much for the great product.

SiboVG, 2023-06-26:
Hey, thank you for your message. I think the best thing to do is to put design notes in the comment section of your rocket - which kinda represents your design.
[image]

If that's not enough for you, we can keep this feature request open.

creilly27, 2023-06-26:
Hello,

That is just what I was looking for.  Thank you.  I don’t think a new
feature is required, but I do think it isn’t in the first place someone
would look to find it if they wanted to take notes.  I might not be too
smart, but I wasn’t thinking there was anything to edit in the root of the
rocket design tree.  Thanks again.

Connor

SiboVG, 2023-06-26:
> Hello, That is just what I was looking for. Thank you. I don’t think a new feature is required, but I do think it isn’t in the first place someone would look to find it if they wanted to take notes. [...]

Happy to help! I admit it's not too obvious. Maybe we should open the rocket editor when a new file is created. I'll relay this with the rest of the team.

closed 2023-06-26 by SiboVG