openrocket/openrocket #2235: [Bug] File incompatibility with OR 15.03 when staging

ReinhardR, 2023-06-26:
### Describe the bug

Existing files created with OR 15.03 break in a minor way when multi-stage rockets are configured for staging at "Launch plus x seconds", after that feature was removed. See #1611. 

The first sign of an issue is the error message: "Invalid parameter encountered, ignoring." when opening the file in OR 22.02. There is no hint about the nature of the parameter.

The simulation will fall back to staging at "Current stage ejection charge plus x seconds", which will inhibit staging on motors without ejection charge. On a cursory look, the simulation runs seemingly fine, but the performance of the rocket will be reduced by dragging along the lower stage.
Examination of the plots will show that lines don't bifurcate (if one pays attention to that).

Two potential ways to resolve this come to mind:
1. Advise user during import of legacy files about the deprecated feature and how the user can fix it.
2. Undo or amend #1860.  "Launch plus x seconds" is not a useless staging configuration as long as x is not zero. In some rare scenarios it's probably the most natural fit (e.g. 3+ stages with simple avionics that don't count number of motor burns).

### To Reproduce

1. Create two-stage rocket in OR 15.03
2. Use plugged motor in booster
3. Set stage separation to "Launch plus 5 seconds"
4. Save file and open in OR 22.02

### OpenRocket version

22.02

### What platform are you running on?

Windows

JoePfeiffer, 2023-06-26:
Oops, you're right.

SiboVG, 2023-06-27:
Just recently there was also a user on the r/rocketry Discord server that could've benefitted from the "Launch plus x seconds" separation event. So I'm okay with reverting the event removal.

PR #2238 (2023-06-27): Revert "Disallow stage separation at launch"
Reverts openrocket/openrocket#1860. Fixes #2235.
closed 2023-06-27 by JoePfeiffer