openrocket/openrocket #2254: Tracking openrocket in downstream package ecosystem

luzpaz, 2023-07-07:
I opened this ticket in the hope of being able to increase the exposure of openrocket. Perhaps there are others who would like to see this as well and contribute their time to help make it happen ?

Here is the current status of openrocket in the package ecosystem: 

[image]

SiboVG, 2023-07-08:
Really nice overview and good idea to update those packages!

SiboVG, 2023-07-17:
I just created a Wikipedia page, which is under review to be published.

luzpaz, 2023-07-18:
Opened a gentoo new package request (https://bugs.gentoo.org/910466)

SiboVG, 2023-07-18:
> Opened a gentoo new package request (https://bugs.gentoo.org/910466)

Thanks @luzpaz! I'm working on updating Debian/Ubuntu and creating a chocolatey package.

luzpaz, 2023-07-23:
> I just created a Wikipedia page, which is under review to be published.

@SiboVG has it been approved?

SiboVG, 2023-07-23:
> > I just created a Wikipedia page, which is under review to be published.
> 
> @SiboVG has it been approved?

No, not yet, I'm still ping-ponging in the review stage. I got the following remarks:

---

Well done on creating the draft, and it may potentially meet the relevant requirements (including [WP:GNG](https://en.wikipedia.org/wiki/Wikipedia:GNG)) but presently does not. As other reviewers have noted, Wikipedia's basic requirement for entry is that the subject is notable. Essentially subjects are presumed notable if they have received significant coverage in multiple published [secondary sources](https://en.wikipedia.org/wiki/Wikipedia:No_original_research#Primary,_secondary_and_tertiary_sources) that are [reliable](https://en.wikipedia.org/wiki/Wikipedia:Reliable_sources), intellectually independent of each other, and [independent of the subject](https://en.wikipedia.org/wiki/Wikipedia:Independent_sources). To properly create such a draft page, please see [referencing for beginners](https://en.wikipedia.org/wiki/Help:Referencing_for_beginners) or the article [Easier Referencing for Beginners](https://en.wikipedia.org/wiki/Wikipedia:ERB). Also, if you have any connection to the subject, including being paid, you have a [conflict of interest](https://en.wikipedia.org/wiki/Wikipedia:COI) that you must declare on your Talk page (to see instructions on how to do this please click the link). Please familiarise yourself with these pages before amending the draft. If you feel you can meet these requirements then resubmit the page and ping me and I would be happy to reassess.

In summary, the draft needs multiple published sources that are:
[in-depth](https://en.wikipedia.org/wiki/Wikipedia:Notability#General_notability_guideline) (not just passing mentions about the subject)
[reliable](https://en.wikipedia.org/wiki/Wikipedia:Reliable_sources)
[secondary](https://en.wikipedia.org/wiki/Wikipedia:No_original_research#Secondary)
[independent](https://en.wikipedia.org/wiki/Wikipedia:Independent_sources) of the subject

---

So we need some more reliable papers that show the importance of OpenRocket...

Here is my draft so far.
[Draft_OpenRocket.pdf](https://github.com/openrocket/openrocket/files/12138511/Draft_OpenRocket.pdf)

SiboVG, 2023-07-23:
@luzpaz I'm gonna put the Debian/Ubuntu and chocolatey package updates on hold to work on some other stuff. If you'd like to take this over, I'd appreciate it!

RuboGubo, 2023-10-12:
Is flatpak going to be supported?

luzpaz, 2023-10-12:
Chocolatey is distributing openrocket :+1:

SiboVG, 2023-10-12:
> Is flatpak going to be supported?

There is an existing request for it: #1697.

SiboVG, 2023-10-12:
> Chocolatey is distributing openrocket 👍

Awesome, thanks! Maybe it's useful to document on our GitHub wiki how we can update the choco OR version, for future releases?

luzpaz, 2023-10-12:
> Awesome, thanks! Maybe it's useful to document on our GitHub wiki how we can update the choco OR version, for future releases?

https://docs.chocolatey.org/en-us/community-repository/users/package-triage-process#package-is-outdated

Here's the chocolatey package page for OR, https://community.chocolatey.org/packages/openrocket#install

SiboVG, 2023-10-12:
> https://docs.chocolatey.org/en-us/community-repository/users/package-triage-process#package-is-outdated
>
> Here's the chocolatey package page for OR, [...]

Great, thanks a lot!

SiboVG, 2026-08-18:
> No, not yet, I'm still ping-ponging in the review stage. I got the following remarks:
>
> Well done on creating the draft, and it may potentially meet the relevant requirements (including [...]

I updated the OpenRocket submission to include more references, and we got accepted: https://wikipedia.org/wiki/OpenRocket

luzpaz, 2026-08-20:
I bumped the wikidata page (https://www.wikidata.org/wiki/Q130573024) so it would reflect the updated stable version in the repology badge when repology gets updated.