openrocket/openrocket #2255: [Bug] un-readable text

Bob-The-Ai, 2023-07-09:
### Describe the bug

The values in the drag characteristics tab of the component analysis interface is white on beige which makes it unreadable.

### To Reproduce

1. Create a basic rocket
2. Tools > Component analysis
3. Go to drag characteristics

### Screenshots / .ork file

[image]

### OpenRocket version

22.02

### What platform are you running on?

Linux

### Additional context

I am using KDE in dark mode.

JoePfeiffer, 2023-07-09:
If you switch to light mode, is it readable (I'm wondering if Java is picking up that you're in dark mode and setting all the text to white in spite of being on a light background.  That's what it looks like to me; having the text become black if you switch to light mode would help confirm it).

JoePfeiffer, 2023-07-09:
We basically don't support dark at the moment.  There is an open PR (#1089) to support it in a future release. Some things work just because Java itself support dark mode, but that's pretty much by accident.

closed 2023-07-09 by JoePfeiffer

Bob-The-Ai, 2023-07-09:
It would appear that the cells not changing color because the color of the cells is "manually" set to indicate how "much" drag there is.

JoePfeiffer, 2023-07-09:
> It would appear that the cells are not changing color because the color of the cells is "manually" set to indicate how "much" drag there is.

Yes, you're correct.  So fixing it would require coming up with a different set of calculations to set the colors when in dark mode.