openrocket/openrocket #2263: [Feature Request] UI - Disable components

stevesaner, 2023-07-17:
In the rocket design tool, I would love to be able to enable and disable individual components that are part of the design. The default of a component should be that it is enabled. But when a component is disabled it should disappear from the rocket design schematic, should be grayed out in the component tree, and all of the calculations, such as total mass, CG, CP, etc. should ignore that component. 

When modeling a rocket, I like to be as granular as possible and include as many details as possible, such as parachute, shock cords, quick links, electronics, etc. But then, I find that I would like to see what the mass should be with out all of those components so that I can compare that with the mass of the actual, empty, rocket. There is currently no way that I know of to do that without deleting the extra components and then having to add and configure them again later.

A satisfactory mechanism would be to disable or enable each component individually, such as in a right-click menu. The ability to select muliple components at once and disable or enable all at once would be great. Another interesting architecture might be to be able to group individual components into a named group and then have the ability disable or enable the group. Component groups could perhaps be useful for other purposes too.

JoePfeiffer, 2023-07-18:
How would disabling and enabling component subtrees work for you?

There have been a number of suggestions involving doing interesting things with subtrees -- read and write them so people could have parts libraries with things like standard e-bays and motor mounts they use, export a subtree as a .OBJ file for 3D printing...  I'm wondering if this is another one.

stevesaner, 2023-07-18:
Maybe a subtree would work, but I think I would prefer a bit more granularity. I'm looking at the tree of a custom design. Part of it is:

Forward Airframe
  - Main Parachute
  - Quick Link
  - Shock cord
  - Quick Link
  - Electronics Bay
      - Forward shock cord mount
      - Forward bulkhead
      - Electronics
      - Aft bulkhead

To do what I would want, I would disable the Main Parachute, both Quick Links, Shock cord, and then maybe the electronics from inside the Electronics bay. The rest I would want to leave enabled.

Now, if one of the components that I could include would be a container for other components and that forms a new sub-tree, then perhaps enable/disable on an entire sub-tree would work.

JoePfeiffer, 2023-07-18:
Subtrees are meant to act as containers, like you suggest.  In some cases it's enforced (if you're building a two stage rocket, the only way to get it to simulate properly is to have all the components in each stage be in the subtree rooted at that stage), in most it isn't and people tend to not make enough use of it.  So my typical e-bay has a switch band, child of the switch band is the coupler tube, children of the coupler tube are the bulkplates and a mass element for the electronics and hardware.

So if you were evaluating two switch band designs, you could have both of them in the design and just enable the one you want.  Likewise things like which of two parachutes you want....

I guess I see it as an extension of the current enable/disable a stage....

JoePfeiffer, 2023-08-11:
Addendum in the light of a recent discord conversation:  if we could have components enabled or disabled depending on what stages are active, we could simulate things like tailcones on upper stages that nest into an interstage.

neilweinstock, 2023-09-06:
See also #2000.

As for subtrees: this seems simple to me.  If you disable the parent object, then all children are automatically disabled as well.  Nothing else would make sense.

You would also be free to individually enable/disable children rather than operating on the whole subtree.