openrocket/openrocket #2264: Rocket active (roll, pitch, yaw) stabilization and different style Airfoil Fin creation

SupSonAlv, 2023-07-19:
-Currently there is only roll stabilization;
     If active stabilization feature can be added,(,PID values,servo shaft 
     load torque,servo sample time,Controller sample time,) features 
      like this it would be perfect and it could appeal to more users.
-Currently, only flat air profiles can draw wings;
    Instead, with different airfoils, especially for supersonic speeds.
     Ailerons can be created.

I know many hobby rocketers who want to do active stabilization. For the most part, manually setting the Required PID Values to do this stabilization is considered a problem that requires a lot of time and expense. What is the probability of adding these two features?

SiboVG, 2023-07-19:
For now, you could write your own scripts for pitch & yaw stabilization. (Edit a simulation -> Simulation options -> Add extension -> Scripts). Building this natively in OR is a low priority for us, so I don't see it happening any time soon, unless we can get some external help on this.

[AeroVECTOR](https://github.com/GuidodiPasquo/AeroVECTOR) is a good program for what you're asking (I see you've already posted something in this repo so you're aware of it, but leaving this reference for other users).

JoePfeiffer, 2023-07-19:
As a core part of OR, I don't see active stabilization ever happening -- while there are people using it, especially for extreme altitude attempts, it's a really tiny niche.  It sounds like you've come across the active roll control plugin; that could be used as a starting point for someone who wanted to write a plugin to do more complete active stabilization. If we got a PR to add that, we'd look at it.

More detailed modelling of airfoils would be a great addition especially for, as you say, supersonic flight.  I don't really know the extent to which it would affect stability vs. just drag; given a good characterization of how they behave (I suspect that's out there, but haven't looked) adding them to our models wouldn't be out of the question.