openrocket/openrocket #2293: Bug Report

Casiofx100, 2023-08-16:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. OpenRocket had file open in the background.
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException at net.sf.openrocket.gui.dialogs.optimization.SimulationModifierTree$ComponentModifierTreeRenderer.getTreeCellRendererComponent(SimulationModifierTree.java:153) at java.desktop/javax.swing.plaf.basic.BasicTreeUI$NodeDimensionsHandler.getNodeDimensions(BasicTreeUI.java:3155) at java.desktop/javax.swing.tree.AbstractLayoutCache.getNodeDimensions(AbstractLayoutCache.java:493) at java.desktop/javax.swing.tree.VariableHeightLayoutCache$TreeStateNode.updatePreferredSize(VariableHeightLayoutCache.java:1344) at java.desktop/javax.swing.tree.VariableHeightLayoutCache.rebuild(VariableHeightLayoutCache.java:723) at java.desktop/javax.swing.tree.VariableHeightLayoutCache.setModel(VariableHeightLayoutCache.java:111) at java.desktop/javax.swing.plaf.basic.BasicTreeUI.configureLayoutCache(BasicTreeUI.java:2103) at java.desktop/javax.swing.plaf.basic.BasicTreeUI.completeUIInstall(BasicTreeUI.java:812) at java.desktop/javax.swing.plaf.basic.BasicTreeUI.installUI(BasicTreeUI.java:772) at java.desktop/javax.swing.JComponent.setUI(JComponent.java:685) at java.desktop/javax.swing.JTree.setUI(JTree.java:706) at java.desktop/javax.swing.JTree.updateUI(JTree.java:727) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1363) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI0(SwingUtilities.java:1378) at java.desktop/javax.swing.SwingUtilities.updateComponentTreeUI(SwingUtilities.java:1354) at java.desktop/sun.swing.plaf.DesktopProperty.updateWindowUI(DesktopProperty.java:124) at java.desktop/sun.swing.plaf.DesktopProperty.updateWindowUI(DesktopProperty.java:127) at java.desktop/sun.swing.plaf.DesktopProperty.updateAllUIs(DesktopProperty.java:116) at java.desktop/com.sun.java.swing.plaf.windows.WindowsDesktopProperty.updateAllUIs(WindowsDesktopProperty.java:52) at java.desktop/sun.swing.plaf.DesktopProperty$1.run(DesktopProperty.java:217) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715) at java.base/java.security.AccessController.doPrivileged(Native Method) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
User-defined thrust curves location: C:\Users\ceymo\AppData\Roaming\OpenRocket\ThrustCurves
JOGL version: 2.4.0-rc-20210111
Current default locale: en_AU
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=c:_program_files_openrocket_openrocket.exe1
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\ceymo\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Windows\System32\OpenSSH\;C:\Users\ceymo\AppData\Local\Microsoft\WindowsApps
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.17+7-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.17
java.version.date=2022-10-18
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.17+7-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln739228102880271898
[... 263 lines cut ...]
45463 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/com.sun.java.swing.plaf.windows.WindowsDesktopProperty.updateAllUIs(WindowsDesktopProperty.java:52)
45464 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/sun.swing.plaf.DesktopProperty$1.run(DesktopProperty.java:217)
45465 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
45466 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:770)
45467 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
45468 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
45469 73067.797 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
45470 73067.797 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
45471 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:740)
45472 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
45473 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
45474 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
45475 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
45476 73067.797 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
[... cut ...]
45483 74927.958 WARN (DoubleModel.java:1006) DoubleModel[constant=0.2] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@260c862c]
45484 74927.958 WARN (DoubleModel.java:1006) DoubleModel[constant=0.0] being garbage-collected while having listeners [net.sf.openrocket.gui.adaptors.DoubleModel$ValueSliderModel@3f80b44b]
45485 103959.227 USER (SwingExceptionHandler.java:220) User requested sending bug report
45486 103959.254 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2023-08-17:
Hi, thanks for your report. Could you please tell us what you were doing that caused the error?

SiboVG, 2023-08-27:
No response from user, closing...

closed 2023-08-27 by SiboVG