openrocket/openrocket #2348: [Bug] Bad specs on pro29 CTI H255-4G motor

wiltomdus, 2023-09-22:
### Describe the bug

The CTI 229H255-14A motor has the wrong specs in the motor configuration

The name of the motor is ok, but the details panel show different values

View screenshots for comparison with Thrustcurve.org data and CTI datasheet

### To Reproduce

1. add a new configuration in the "motor & configuration" section
2. Search for 229H255-14A
3. Select 229H255-14A and navigate to the "Show details" tab
4. Compare specs to the CTI datasheet (www.pro38.com/products/pro29/motor/MotorData.php?prodid=229H255-14A)

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[CTI-229H255-14A.zip](https://github.com/openrocket/openrocket/files/12705137/CTI-229H255-14A.zip)

### OpenRocket version

22.02

### What platform are you running on?

Windows

JoePfeiffer, 2023-09-23:
We get all our thrust curves directly from thrustcurve.org. They appear to have two thrust curves for this motor, one which matches CTI's spec sheet and one which does not; both were entered by the same person, both are stated to be from Tripoli certification testing. It's troubling that one of them is so much "beefier" than the other, but it's out of our control. I wouldn't be at all surprised to learn these are just test firings of two motors, though the difference between them seems quite a bit greater than most.

Anyway, if you select the motor and then go to the dropdown box at the top left of the dialog, you can select the "blue" motor. This is the one that matches CTI's documentation.

closed 2023-09-23 by JoePfeiffer

michaelsheffernan, 2026-10-11:
hey, found the root cause on this one while debugging our own sims.

the second 229H255-14A curve isn't a second test firing. ThrustCurve's RockSim file for it (simfile 5f4294d20002e900000005ee) was a byte-for-byte copy of the 315H255-14A file: H255-BS, 316 Ns, 1.25 s burn.

we caught it because OpenRocket said our rocket would hit Mach 1.7 and the rocket equation said that was impossible.

reported it to ThrustCurve on 11 Oct 2026, they pulled the file same day. OpenRocket 24.12's bundled motor db still has it until the next update though.

workaround for now: in the motor dialog's thrust-curve dropdown, pick the RASP curve (229 Ns, 0.94 s burn).

cheers