openrocket/openrocket #2351: Bug when viewing in Pop OS

erobinson-github-io, 2023-09-25:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalStateException: getRocket() called with root component Stage
at net.sf.openrocket.rocketcomponent.RocketComponent.getRocket(RocketComponent.java:2006)
at net.sf.openrocket.rocketcomponent.RocketComponent.addComponentChangeListener(RocketComponent.java:2199)
at net.sf.openrocket.gui.adaptors.PresetModel.<init>(PresetModel.java:45)
at net.sf.openrocket.gui.configdialog.RocketComponentConfig.<init>(RocketComponentConfig.java:146)
at net.sf.openrocket.gui.configdialog.NoseConeConfig.<init>(NoseConeConfig.java:52)
at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance0(Native Method)
at java.base/jdk.internal.reflect.NativeConstructorAccessorImpl.newInstance(NativeConstructorAccessorImpl.java:62)
at java.base/jdk.internal.reflect.DelegatingConstructorAccessorImpl.newInstance(DelegatingConstructorAccessorImpl.java:45)
at java.base/java.lang.reflect.Constructor.newInstance(Constructor.java:490)
at net.sf.openrocket.gui.configdialog.ComponentConfigDialog.getDialogContents(ComponentConfigDialog.java:160)
at net.sf.openrocket.gui.configdialog.ComponentConfigDialog.setComponent(ComponentConfigDialog.java:108)
at net.sf.openrocket.gui.configdialog.ComponentConfigDialog.<init>(ComponentConfigDialog.java:61)
at net.sf.openrocket.gui.configdialog.ComponentConfigDialog.showDialog(ComponentConfigDialog.java:268)
at net.sf.openrocket.gui.main.ComponentAddButtons$ComponentButton.fireActionPerformed(ComponentAddButtons.java:491)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2308)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
at java.desktop/java.awt.Component.processMouseEvent(Component.java:6635)
at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3342)
at java.desktop/java.awt.Component.processEvent(Component.java:6400)
at java.desktop/java.awt.Container.processEvent(Container.java:2263)
at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5011)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2321)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4918)
at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4547)
at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4488)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
at java.base/java.security.AccessController.doPrivileged(Native Method)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)

---------- Thread information ----------
Thread[AWT-EventQueue-0,6,main]

---------- System information ----------
OpenRocket version: 22.02
[... 456 lines cut ...]
23474 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2307)
23475 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2772)
23476 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Component.dispatchEvent(Component.java:4843)
23477 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:772)
23478 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:721)
23479 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:715)
23480 504.346 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
23481 504.346 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
23482 504.346 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:95)
23483 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:745)
23484 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:743)
23485 504.346 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(Native Method)
23486 504.346 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:85)
23487 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:742)
23488 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
23489 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
23490 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
23491 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
23492 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
23493 504.346 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
23494 504.346 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
23495 504.346 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
23496 514.267 USER (SwingExceptionHandler.java:220) User requested sending bug report
23497 514.267 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2023-09-25:
Thanks for your report -- please give us some more information about how we can reproduce the error. So we really need something like "1. open the attached .ork file, 2. click on... "

SiboVG, 2023-11-22:
Lack of information from user, closing.

closed 2023-11-22 by SiboVG