openrocket/openrocket #2353: Optimum Delay column of Flight Simulations tab is not populated

Roo59, 2023-09-27:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.Design created with this version
2.on the Flight simulations tab, the Optimum Delay column is not populated.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- System information ----------
OpenRocket version: 23.09.beta.01
OpenRocket source: default
OpenRocket location: C:\Program Files\OpenRocket\.\OpenRocket.jar
User-defined thrust curves location: C:\Users\alexl\AppData\Roaming\OpenRocket\ThrustCurves
LAF: com.sun.java.swing.plaf.windows.WindowsLookAndFeel
JOGL version: 2.5.0-rc-20230523
Current default locale: en_AU
System properties:
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.moduleName=C:\Program Files\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\c:_program_files_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=C:\Program Files\OpenRocket\
install4j.exeDir=C:\Program Files\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.class.path=C:\Program Files\OpenRocket\.install4j\i4jruntime.jar;C:\Program Files\OpenRocket\.\OpenRocket.jar
java.class.version=61.0
java.home=c:\program files\openrocket\jre
java.io.tmpdir=C:\Users\alexl\AppData\Local\Temp\
java.library.path=c:\program files\openrocket\jre\bin;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\Program Files\Microchip\xc8\v2.41\bin;C:\Program Files\Microchip\xc16\v2.00\bin;C:\Program Files\Docker\Docker\resources\bin;C:\Users\alexl\AppData\Local\Microsoft\WindowsApps;
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=17.0.7+7-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=17
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=17.0.7
java.version.date=2023-04-18
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=17
java.vm.vendor=BellSoft
java.vm.version=17.0.7+7-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln17771990699299607277
line.separator=\u000d\u000a
native.encoding=Cp1252
os.arch=amd64
os.name=Windows 10
os.version=10.0
path.separator=;
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
[... 128 lines cut ...]
281 32.517 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0779579 user scale, 441.966 overall scale
282 52.143 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0779579 user scale, 441.966 overall scale
283 52.144 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0779579 user scale, 441.966 overall scale
284 61.276 DEBUG (FileHelper.java:144) File name does not contain extension, adding 'obj'
285 61.285 VBOSE (DoubleModel.java:953) DoubleModel[OBJExportOptions:ScalingDouble] adding listener (total 1): javax.swing.JSpinner$ModelListener@2bc0a17d
286 61.286 VBOSE (DoubleModel.java:953) DoubleModel[OBJExportOptions:ScalingDouble] adding listener (total 2): net.sf.openrocket.file.wavefrontobj.OBJOptionChooser$10@3b9901ad
287 61.287 VBOSE (DoubleModel.java:953) DoubleModel[OBJExportOptions:ScalingDouble] adding listener (total 3): net.sf.openrocket.file.wavefrontobj.OBJOptionChooser$7@17386cd0
288 61.292 DEBUG (DoubleModel.java:809) Setting value 1000.0 for DoubleModel[OBJExportOptions:ScalingDouble]
289 103.210 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0779579 user scale, 441.966 overall scale
290 103.213 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0779579 user scale, 441.966 overall scale
291 103.304 WARN (DoubleModel.java:1006) DoubleModel[OBJExportOptions:ScalingDouble] being garbage-collected while having listeners [javax.swing.JSpinner$ModelListener@2bc0a17d, net.sf.openrocket.file.wavefrontobj.OBJOptionChooser$10@3b9901ad, net.sf.openrocket.file.wavefrontobj.OBJOptionChooser$7@17386cd0]
292 105.086 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0779579 user scale, 441.966 overall scale
293 143.594 DEBUG (GUIUtil.java:381) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1295,height=695]
294 143.595 DEBUG (GUIUtil.java:386) Storing maximized state of net.sf.openrocket.gui.main.BasicFrame
295 143.677 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0882380 user scale, 500.247 overall scale
296 143.678 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0882380 user scale, 500.247 overall scale
297 143.679 VBOSE (AbstractScaleFigure.java:193) ____ Updating RocketFigure to: 0.0882380 user scale, 500.247 overall scale
298 243.622 DEBUG (GUIUtil.java:381) Storing size of net.sf.openrocket.gui.main.BasicFrame: java.awt.Dimension[width=1295,height=695]
299 243.623 DEBUG (GUIUtil.java:386) Storing maximized state of net.sf.openrocket.gui.main.BasicFrame
300 248.944 USER (BasicFrame.java:796) About selected
301 251.885 DEBUG (GUIUtil.java:272) Clearing all models of window net.sf.openrocket.gui.dialogs.AboutDialog[dialog0,127,127,453x490,invalid,hidden,layout=java.awt.BorderLayout,APPLICATION_MODAL,title=OpenRocket 23.09.beta.01,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,7,30,439x453,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
302 251.887 DEBUG (MemoryManagement.java:56) Adding object into collectable list: net.sf.openrocket.gui.dialogs.AboutDialog[dialog0,127,127,453x490,invalid,hidden,layout=java.awt.BorderLayout,APPLICATION_MODAL,title=OpenRocket 23.09.beta.01,defaultCloseOperation=DISPOSE_ON_CLOSE,rootPane=javax.swing.JRootPane[,7,30,439x453,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
303 263.463 USER (BasicFrame.java:754) Bug report selected
304 263.465 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

SiboVG, 2023-09-27:
Hi, thanks for your report. Could you please share the .ork file that produces this error?

Roo59, 2023-09-28:
Sorry,  I thought that I had attached it.

It is attached on this email

Regards,
Richard.

Roo59, 2023-09-28:
> Hey Richard, unfortunately file attachments from GitHub mails don't end up
> here. You'll have to go to this issue's GitHub page and drag-and-drop your
> file here in a comment. GitHub doesn't allow .ork attachments, so you'll
> have zip the file first.

Thanks, I'll check it out!

PR #2363 (2023-10-09): Fix 2353
Instead of saving optimum delay in a simulations flight summary data, it was recalculated every time it was displayed. So, only saving flight summary data didn't save the optimal delay, and with no flight data branches saved in this circumstance it couldn't be recalculated. This puts it in the flight summary data, and saves it to and loads it from a .ork file

Fixes #2353
closed 2023-10-09 by JoePfeiffer