openrocket/openrocket #2371: [Bug] Stiffener inside Coupler = no 3D view for Figure, Unfinished, Finished

jimmiedave, 2023-10-16:
### Describe the bug

Creating a model of a new Mega Der Red Max I'm building. I've got a coupler with a stiffener inside for robustness where the top tube will be removable. Trying to switch to 3D Finished view, I get a black View window, no model. Initially got a white View window. Black started on quitting and restarting OR.

### To Reproduce

1. Open enclosed .ork (on Mac OS Catalina, Mac Book Pro 16in Intel 2019, Radeon graphics, on external monitor)
2. Try to switch View Type to "3D Finished"
3. Observe black View window
4. Delete one or both stiffener and coupler
5. Try to switch View Type to "3D Finished" again
6. Note Finished View appears.
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*
[image]

[image]

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[Daves 38mm Mega Der Red Max_or22ß_231015.ork.zip](https://github.com/openrocket/openrocket/files/12911657/Daves.38mm.Mega.Der.Red.Max_or22ss_231015.ork.zip)

### OpenRocket version

22.09 beta (from Drive)

### What platform are you running on?

macOS

jimmiedave, 2023-10-16:
OK, maybe the bug is with the 7.5in LOC Coupler Stiffener I pulled from the Parts Library

I used the LOC part STC-3.90, and shortened it up to 140mm as was my need.

I suffered the 3D problem as listed above.

I re-opened the .ork file, deleted the stiffener, and added it back using the same settings by hand, and tested: No more 3D problem.

Changing issue title.

JoePfeiffer, 2023-10-17:
Not seeing the issue under Linux. I can open the .ork and  view it in all three 3D views in the design window, and in the photo studio.

SiboVG, 2023-10-17:
Also no issues on MacBook Pro M1, macOS 14. Strange...

jimmiedave, 2023-10-17:
Interesting. Same issue came back for me immediately on the file next day after a cold start.

Suggestion: I'll try and recreate the problem in a new file with the part from parts library, and if no issue, we call it "random cosmic ray" or something.

jimmiedave, 2023-10-22:
Windows 7 VM, same Mac, upon switching to one of the 3D views (don't think this is a valid part of the issue, but for a full report):

[image]

jimmiedave, 2023-10-22:
Not seeing on Ubuntu Linux, different iron.

jimmiedave, 2023-10-22:
Tried to recreate with the changed JAR in the Drive files. Cannot break it in the way the weird file does. Closing.

closed 2023-10-22 by jimmiedave