openrocket/openrocket #2373: [Bug] Auto pop of parts library resets units (to prior value?)

jimmiedave, 2023-10-16:
### Describe the bug

Noticed during testing that my units kept reverting from mm to cm. The parts library was set to "Always open...".

Figured out that the pop up of the parts library appeared to be causing the trouble.

### To Reproduce

Setup:

1. Set parts library to "Always open..." when adding a body tube
2. Close any design

Demo:

1. Set Units:Rocket dimensions to "cm" and "OK"
2. New design
3. Click sustainer
4. Add a body tube
5. Close Parts Library and Cancel adding a body tube
6. Set Units:Rocket dimensions to "mm" and OK
7. Quit OpenRocket
8. Start OpenRocket
9. Click sustainer
10. Add a body tube
11. **BUG:** Note rocket dimensions are again in "cm"

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

22.09 from Drive

### What platform are you running on?

macOS

SiboVG, 2023-10-21:
I'm not seeing this bug on my machine... Can someone else confirm? You're quite good at triggering custom bugs @jimmiedave 😉.

jimmiedave, 2023-10-21:
Is it possible that I've got a permissions problem with preferences on my machine, or something similar?

jimmiedave, 2023-10-22:
On my Catalina Intel Mac, in `~/Preferences/`, I see:

`./openrocket.favoritepresets.centering_ring.plist
./openrocket.favoritepresets.nose_cone.plist
./openrocket.favoritepresets.bulk_head.plist
./openrocket.favoritepresets.launch_lug.plist
./openrocket.favoritepresets.rail_button.plist
./openrocket.favoritepresets.body_tube.plist
./openrocket.favoritepresets.engine_block.plist
./openrocket.favoritepresets.tube_coupler.plist
./openrocket.favoritepresets.streamer.plist
./openrocket.objexportoptions.coordtransform.plist
./openrocket.favoritepresets.transition.plist
./openrocket.favoritepresets.parachute.plist`

Where are OR's actual prefs (to include Units) saved on MacOS, that I might check and see if I have a permissions issue? Doesn't seem like it'd be any of these.

jimmiedave, 2023-10-22:
Not seeing reset of prefs in Ubuntu Linux

jimmiedave, 2023-10-22:
OK, kind of mean trick to switch out the .jar file in Drive...

Testing with NOT the file I've been testing with since they were released to Drive, I no longer see units being reset.

jimmiedave, 2023-10-22:
OK, can't see this, and it's apparent the JAR in Drive has changed. Closing.

closed 2023-10-22 by jimmiedave