openrocket/openrocket #2378: [Bug] Nose Cone Shoulder changes diameter between save and re-open

jimmiedave, 2023-10-20:
### Describe the bug

I've repeatedly re-set the shoulder of my nose cone to 100mm diameter.

Whenever I save the model, upon re-opening it, the shoulder has changed to a much smaller diameter. This time (first time I've looked at the exact number in the diameter field) it's 50mm in diameter.

Before Save:

[image]

After quit, restart OR, Load:

[image]

fixed shoulder, saved again

After OR Close File, Open file (very careful to open exactly the file I've just saved this time):

[image]

I expect the nose cone shoulder to stay at the diameter I've set it.

### To Reproduce

1. Open enclosed .ork file
2. Note nose cone shoulder is 50mm / 5cm wide
3. Adjust to correct 100mm / 10cm wide
4. Save file
5. Close file
6. Re-open just-saved file
7. **BUG:** Note nose cone has changed to 50mm / 5cm shoulder
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*
See above

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[Daves 38mm Mega Der Red Max_or22ß_231015.ork.zip](https://github.com/openrocket/openrocket/files/13054941/Daves.38mm.Mega.Der.Red.Max_or22ss_231015.ork.zip)

### OpenRocket version

22.09 beta from drive

### What platform are you running on?

macOS

### Additional context

Mac OS Catalina, run as .jar from command line.

Also note that rocket measurement units still reverting to mm, even though I've not accessed the Parts Library (see #2373)

neilweinstock, 2023-10-20:
Very odd.  I can reproduce this with your file, but not on any file I've created on my own.

jimmiedave, 2023-10-20:
I've wondered whether it's the Eggtimer that lives in the shoulder - maybe there's code that's trying to get the shoulder out of the way, but that doesn't seem like it makes any kind of sense. Maybe 'cuz it's an internal tube that's not a motor mount tube, the shoulder thinks it has something to do? Doesn't make any more sense than the first idea.

PR #2379 (2023-10-20): Include radius clamping in clamping exclude
This PR fixes #2378.
closed 2023-10-20 by SiboVG