openrocket/openrocket #2380: [Bug] Jogl can not work on jdk17

ZCY571864845, 2023-10-21:
### Describe the bug

Can not use openrocket figure3d view type on jdk17.
GLException:Unable to determine GraphicsGonfiguration.

### To Reproduce

1. Open or create any rocket.
2. Change view type to 3D Figure.
3. Openrocket throw uncaught exception.

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*
[image]

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

unstable

### What platform are you running on?

Windows

SiboVG, 2023-10-21:
Thank you for your report, this is a duplicate of #2336.

closed 2023-10-21 by SiboVG

ZCY571864845, 2023-10-21:
I add VM arguments:
--add-exports java.base/java.lang=ALL-UNNAMED
--add-exports java.desktop/sun.awt=ALL-UNNAMED
--add-exports java.desktop/sun.java2d=ALL-UNNAMED
and it works.
but,I find it is very hard to select a component on 3d view type.

SiboVG, 2023-10-21:
> I add VM arguments: --add-exports java.base/java.lang=ALL-UNNAMED --add-exports java.desktop/sun.awt=ALL-UNNAMED --add-exports java.desktop/sun.java2d=ALL-UNNAMED and it works. but,I find it is very hard to select a component on 3d view type.

The 3D component selection code is very inefficient, so it can sometimes not react the way you'd want it. If it makes you feel any better, 3D component selection simply doesn't work on macOS 🙃.

ZCY571864845, 2023-10-21:
3D component selection seems work better on version OpenRocket 22.02(jdk11).Maybe jogl do not support very friendly on jdk17.

ZCY571864845, 2023-10-21:
I Change the jdk version to 11,3D component selection also work hard.Obviously,It is not cause by jogl compatibility on jdk17.

SiboVG, 2023-10-21:
> I Change the jdk version to 11,3D component selection also work hard.Obviously,It is not cause by jogl compatibility on jdk17.

Okay, thanks for testing!