openrocket/openrocket #2383: Openrocket file not saving after edits.

kavred, 2023-10-24:
---------- Bug report ----------

1. Like any other day I inserted my usb into my computer to edit my rocket to see if I could fly it higher.
2. I edited a few things like motor configuration, bottom tube lenght, and transition piece weight.
3. Then I tried to save the file but I encountered an error of which there should be a screenshot.

Searched this online, but I couldn't really find a way to solve. 
I would hope that this get's solved quick since I'm on a time crunch.

---------- System information ----------
OpenRocket version: 22.02
OpenRocket source: default
OpenRocket location: E:\Rocketry\Open_Rocket\OpenRocket\.\OpenRocket.jar
User-defined thrust curves location: C:\Users\chess\AppData\Roaming\OpenRocket\ThrustCurves
JOGL version: 2.4.0-rc-20210111
Current default locale: en_US
System properties:
awt.toolkit=sun.awt.windows.WToolkit
exe4j.consoleCodepage=cp0
exe4j.isInstall4j=true
exe4j.launchName=E:\Rocketry\Open_Rocket\OpenRocket\OpenRocket.exe
exe4j.moduleName=E:\Rocketry\Open_Rocket\OpenRocket\OpenRocket.exe
exe4j.semaphoreName=Local\e:_rocketry_open_rocket_openrocket_openrocket.exe0
exe4j.tempDir=
exe4j.unextractedPosition=0
file.encoding=Cp1252
file.separator=\
install4j.appDir=E:\Rocketry\Open_Rocket\OpenRocket\
install4j.exeDir=E:\Rocketry\Open_Rocket\OpenRocket\
install4j.launcherId=59
install4j.swt=false
java.awt.graphicsenv=sun.awt.Win32GraphicsEnvironment
java.awt.printerjob=sun.awt.windows.WPrinterJob
java.class.path=E:\Rocketry\Open_Rocket\OpenRocket\.install4j\i4jruntime.jar;E:\Rocketry\Open_Rocket\OpenRocket\.\OpenRocket.jar
java.class.version=55.0
java.home=e:\rocketry\open_rocket\openrocket\jre
java.io.tmpdir=C:\Users\chess\AppData\Local\Temp\
java.library.path=e:\rocketry\open_rocket\openrocket\jre\bin;C:\Program Files\Python37\Scripts\;C:\Program Files\Python37\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\dotnet\;C:\Users\chess\AppData\Local\Microsoft\WindowsApps;C:\Users\chess\AppData\Local\Programs\Git\cmd
java.runtime.name=OpenJDK Runtime Environment
java.runtime.version=11.0.17+7-LTS
java.specification.name=Java Platform API Specification
java.specification.vendor=Oracle Corporation
java.specification.version=11
java.util.Arrays.useLegacyMergeSort=true
java.vendor=BellSoft
java.vendor.url=https://bell-sw.com/
java.vendor.url.bug=https://bell-sw.com/support
java.version=11.0.17
java.version.date=2022-10-18
java.vm.compressedOopsMode=Zero based
java.vm.info=mixed mode
java.vm.name=OpenJDK 64-Bit Server VM
java.vm.specification.name=Java Virtual Machine Specification
java.vm.specification.vendor=Oracle Corporation
java.vm.specification.version=11
java.vm.vendor=BellSoft
java.vm.version=11.0.17+7-LTS
jdk.debug=release
jnlp.jogamp.tmp.cache.root=jln5979380051146802142
line.separator=\u000d\u000a
os.arch=amd64
os.name=Windows 11
os.version=10.0
path.separator=;
sun.arch.data.model=64
sun.awt.enableExtraMouseButtons=true
sun.awt.exception.handler=net.sf.openrocket.gui.main.SwingExceptionHandler$AwtHandler
sun.boot.library.path=e:\rocketry\open_rocket\openrocket\jre\bin
sun.cpu.endian=little
sun.cpu.isalist=amd64
[... 228 lines cut ...]
69113 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Stage
69114 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Nose Cone
69115 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Body Tube
69116 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Mass Component
69117 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Mass Component
69118 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Mass Component
69119 1020.832 DEBUG (OpenRocketSaver.java:274) Saving component Transition
69120 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Shock Cord
69121 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Body Tube
69122 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Parachute
69123 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Inner Tube
69124 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Centering Ring
69125 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Centering Ring
69126 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Elliptical Fin Set
69127 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Launch Lug
69128 1020.848 DEBUG (OpenRocketSaver.java:274) Saving component Launch Lug
69129 1020.848 DEBUG (OpenRocketSaver.java:431) Saving Photo Settings
69130 1020.848 DEBUG (OpenRocketSaver.java:111) Writing complete, flushing buffers
69131 1020.910 INFO (SwingWorkerDialog.java:151) Worker completed before opening dialog
69132 1020.910 WARN (BasicFrame.java:1553) An I/O error occurred while saving E:\Rocketry\rocket_2.ork
> java.io.EOFException: Unexpected end of ZLIB input stream > at java.base/java.util.zip.InflaterInputStream.fill(InflaterInputStream.java:245) > at java.base/java.util.zip.InflaterInputStream.read(InflaterInputStream.java:159) > at java.base/java.util.zip.ZipInputStream.read(ZipInputStream.java:197) > at java.base/java.io.BufferedInputStream.fill(BufferedInputStream.java:252) > at java.base/java.io.BufferedInputStream.read1(BufferedInputStream.java:292) > at java.base/java.io.BufferedInputStream.read(BufferedInputStream.java:351) > at java.base/java.io.FilterInputStream.read(FilterInputStream.java:107) > at net.sf.openrocket.util.FileUtils.copy(FileUtils.java:25) > at net.sf.openrocket.util.FileUtils.readBytes(FileUtils.java:35) > at net.sf.openrocket.document.attachments.ZipFileAttachment.getBytes(ZipFileAttachment.java:33) > at net.sf.openrocket.document.DecalRegistry$DecalImageImpl.getBytes(DecalRegistry.java:176) > at net.sf.openrocket.file.GeneralRocketSaver.saveAllPartsZipFile(GeneralRocketSaver.java:213) > at net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:185) > at net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:108) > at net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:81) > at net.sf.openrocket.gui.util.SaveFileWorker.doInBackground(SaveFileWorker.java:37) > at net.sf.openrocket.gui.util.SaveFileWorker.doInBackground(SaveFileWorker.java:11) > at java.desktop/javax.swing.SwingWorker$1.call(SwingWorker.java:304) > at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264) > at java.desktop/javax.swing.SwingWorker.run(SwingWorker.java:343) > at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128) > at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628) > at java.base/java.lang.Thread.run(Thread.java:829)
69133 1474.098 USER (BasicFrame.java:781) Bug report selected
69134 1474.098 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
[image]

JoePfeiffer, 2023-10-24:
Is there any chance either your hard drive or your USB stick is full?

kavred, 2023-10-25:
No, both the hard drive and the USB had more than 150 gigabytes of storage left.

JoePfeiffer, 2023-10-25:
Please post the .ork file as it exists before your edits, and exactly what edits can be performed to trigger the bug.

FWIW, the stack trace looks like it's trying to read a decal file to insert it in the .ork, but reading the file is failing.

kavred, 2023-10-29:
Sorry, but I am not able to find the file anymore, I believe I have accidentally deleted it and have also created a new file. The new file is an almost exact copy of the old one with a few different measurements and new names for the parts. I am not facing any errors as of now.

closed 2023-10-29 by kavred