openrocket/openrocket #2384: Adding a Run All Simulations button

kavred, 2023-10-24:
I have quite a few simulations that I have to run after every edit I make and this task seems to be very annoying to me since I have to do it all the time.

I would definitely appreciate a button that just runs all the simulations for me.

Don't really think there are many less time consuming alternatives for this.

JoePfeiffer, 2023-10-24:
You should be able to click the first sim and shift-click the last one, and then click "Run simulations" to get them all run.

kavred, 2023-10-25:
True, but I have more than 300 simulations. (Scrolling does take a second or two.) However if it's not possible, I won't mind.

jimmiedave, 2023-10-25:
CTRL+A (Select All), Hit "Run Simulations" button.

CMD+A if on a Mac for (Select All)

neilweinstock, 2023-10-25:
Yes, CTRL-A works.  Also, if you want to run all simulations after every edit then just enable that in simulation preferences.
[image]

neilweinstock, 2023-10-25:
Request is adequately covered by existing program functionality.  Closing.

closed 2023-10-25 by neilweinstock