openrocket/openrocket #2394: [Bug] Mass and CG overrides have an unexpected interaction

JoePfeiffer, 2023-11-10:
### Describe the bug

Turning CG override for all subcomponents on and off can affect rocket mass.

### To Reproduce

1. Open "simple model rocket" example. Note rocket mass is 48.2g and CG is 23.5cm.
2. Override mass and set to 50g. Note mass is now 98.2g if "override for subcomponents" is off. Leave it off.
3. Override CG and set to 20cm. Note mass is now 98.2 if "override (CG) for subcomponents" and 48.2 if on. It shouldn't change.
4. Note that if mass override is set on for subcomponents then mass remains 50g whether CG override is on or off.
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

unstable

### What platform are you running on?

Linux

### Additional context

This seems a lot like #2393 (Overrides tab is confusing for Stages) but is distinct enough to warrant its own issue, and is definitely a bug.

JoePfeiffer, 2023-11-13:
Ah, it's worse than I thought. In the attached pathologically simple "rocket" (just a nose cone and body tube), adding a mass override to the stage moves the CG.
[nose-body.zip](https://github.com/openrocket/openrocket/files/13330343/nose-body.zip)

SiboVG, 2023-11-13:
Are you already looking into this @JoePfeiffer? I've tried but don't have a lot of time on my hands to debug it thoroughly...

JoePfeiffer, 2023-11-13:
Yes. I was swamped until yesterday afternoon; giving a serous look now.

PR #2397 (2023-11-15): Clean up and fix mass and CG overrides
Corrects some comments involving overrides in RocketComponent (the comments seem to have not been updated when subcomponent overrides for mass and CG were split)
Cleans up and corrects mass and CG override code in MassComputation
Adds a unit test for mass overrides in stages

Fixes #2394
closed 2023-11-15 by JoePfeiffer