openrocket/openrocket #2404: Strange errors trying to plot this design

neilweinstock, 2023-11-17:
This CHAD-staged design yields some weird errors when plotting.
[Estes Big Bertha 3 Stage.zip](https://github.com/openrocket/openrocket/files/13396638/Estes.Big.Bertha.3.Stage.zip)

Here's what the design it looks like:
[image]

The two booster stages just have a body tube, but nothing to create stability.  I'm guessing this is where the problem lies.

Steps to recreate:
1) Start with a fresh instance of 23.09.
2) Load the attached ORK file
3) Go to Flight Simulations, select the first sim, and click "Plot / Export".  The little progress dialog says "Running simulations", and then you get this:
[image]

Although it says "simulation branch "Sustainer"", I guessed that it is after first stage burnout and it is gagging on the finless first stage which has just been ejected.  But this might not be right.
4) Click OK
5) The Plot dialog then comes up.  Click "Plot".  Here's what you get:
[image]

Note that this should normally show additional lines for the two booster stages after separation, but they're not there.  This supports my theory about the finless boosters.  But notice that the Exception actually occurs at the end of the plot, at apogee.  This... is puzzling to me.

Anyway: if the finless boosters are causing the problem, there should be a graceful way of handling them.  May need to make an assumption about stability (assume -1 caliber or something like that?) to enable the plot to finish gracefully.

JoePfeiffer, 2023-11-17:
This one is a regression between 22.02 and the beta. No idea (at the moment) where this behavior would be coming from...

SiboVG, 2024-12-26:
I think this issue is fixed in 24.12.beta.01. I'm not getting an exception and the plot appears logical:
[image]

@neilweinstock or @JoePfeiffer could you please verify?

JoePfeiffer, 2024-12-26:
Yes, this was fixed as part of #2431 and should have been closed then.

closed 2024-12-26 by JoePfeiffer