openrocket/openrocket #2409: [Bug] Nosecone - Shape parameter is approximated when it shouldn't

Exsthain, 2023-11-20:
### Describe the bug

When trying to use a 3/4 parabolic nosecone, the shape parameters get approximated to 0.8

### To Reproduce

1. input 3/4 or .75 in the nosecone shape parameter box

### Screenshots / .ork file

[image]

### OpenRocket version

23.09 stable

### What platform are you running on?

Windows

Exsthain, 2023-11-20:
Shouldn't this be considered a bug as it affects the geometry/simulation of the rocket and it worked just fine in the last stable version?

SiboVG, 2023-11-20:
> Shouldn't this be considered a bug as it affects the geometry/simulation of the rocket and it worked just fine in the last stable version?

Well, whatever it is, I pushed a fix for it in #2411.

Exsthain, 2023-11-21:
> > Shouldn't this be considered a bug as it affects the geometry/simulation of the rocket and it worked just fine in the last stable version?
> 
> Well, whatever it is, I pushed a fix for it in #2411.

Alright, thank you very much!

PR #2411 (2023-11-22): [#2409] Don't use fixed precision for shape parameter
This PR fixes #2409. You can now enter values up to 3 decimals, while the spinner's up- and down arrows increment/decrement in 0.1 steps.

https://github.com/openrocket/openrocket/assets/11031519/417ae0dc-2672-40e7-81b9-7a50ce54d20c
closed 2023-11-22 by SiboVG