openrocket/openrocket #2419: List of 3D view improvements

SiboVG, 2023-12-06:
This issue is meant to keep track of which features are currently missing in the 3D view:

- [ ] Fin cross-section
- [ ] Fin fillets
- [ ] Rendering of cluster mounts, see #137
- [ ] Roughness - we should display the roughness of the rocket component using some sort of roughness bump map

Please leave a reply in case I missed something.

Aeroengineer1, 2024-04-20:
Adding thoughts to this, though could increase complexity.  For airfoil shapes, it would be great if there was a feature that allowed you to specify the actual airfoil at a specific span location.  You might have a simple version that allowed a dropdown that allowed you to select between a certain group of airfoils that are preloaded or load your own points to create the airfoil.  The second version could be like what is found in OpenVSP or XFLR5 where you can define different airfoils at different stations and different overall wing configurations with it.

SiboVG, 2024-04-24:
> Adding thoughts to this, though could increase complexity. For airfoil shapes, it would be great if there was a feature that allowed you to specify the actual airfoil at a specific span location. [...]

Good suggestion, thanks! But I think this needs to be a separate issue. We would first need to incorporate this in the aerodynamics calculations. There's no use to having a nice 3D visualization if the calculations don't back it up :). @Aeroengineer1 could you create a new issue for this please?

Aeroengineer1, 2024-05-27:
@SiboVG sorry somehow missed this.  I will create a new issue and link back to this one.

SiboVG, 2024-05-27:
> @SiboVG sorry somehow missed this. I will create a new issue and link back to this one.

No problem, thanks!