openrocket/openrocket #2422: [Bug] NPE - Cannot read field "Y" because "rect" is null

nevinsb, 2023-12-13:
### Describe the bug

---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1.Edit EB mass component to center 0 in
2. Drag EB mass component to center body tube
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

nevinsb@aol.com  I will include ORK file

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.NullPointerException: Cannot read field "y" because "rect" is null
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.getDropLineRect(BasicTreeUI.java:1541)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.paintDropLine(BasicTreeUI.java:1477)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.paint(BasicTreeUI.java:1441)
at java.desktop/javax.swing.plaf.ComponentUI.update(ComponentUI.java:161)
at java.desktop/javax.swing.JComponent.paintComponent(JComponent.java:842)
at java.desktop/javax.swing.JComponent.paint(JComponent.java:1119)
at java.desktop/javax.swing.JComponent.paintToOffscreen(JComponent.java:5311)
at java.desktop/javax.swing.BufferStrategyPaintManager.paint(BufferStrategyPaintManager.java:246)
at java.desktop/javax.swing.RepaintManager.paint(RepaintManager.java:1337)
at java.desktop/javax.swing.JComponent._paintImmediately(JComponent.java:5259)
at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5069)
at java.desktop/javax.swing.JComponent.paintImmediately(JComponent.java:5081)
at com.github.weisj.darklaf.ui.tree.DarkTreeExpansionAnimationListener$TreeStateAnimator.repaint(DarkTreeExpansionAnimationListener.java:95)
at com.github.weisj.darklaf.ui.tree.DarkTreeExpansionAnimationListener$TreeStateAnimator.onAnimationFinished(DarkTreeExpansionAnimationListener.java:108)
at com.github.weisj.darklaf.graphics.Animator.stop(Animator.java:297)
at com.github.weisj.darklaf.ui.tree.DarkTreeExpansionAnimationListener.startAnimation(DarkTreeExpansionAnimationListener.java:68)
at com.github.weisj.darklaf.ui.tree.DarkTreeExpansionAnimationListener.treeExpanded(DarkTreeExpansionAnimationListener.java:55)
at java.desktop/javax.swing.JTree.fireTreeExpanded(JTree.java:2864)
at java.desktop/javax.swing.JTree.setExpandedState(JTree.java:3766)
at java.desktop/javax.swing.JTree.expandPath(JTree.java:2296)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:99)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:102)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:102)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:102)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:91)
at net.sf.openrocket.gui.components.BasicTree.treeDidChange(BasicTree.java:77)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.updateSize(BasicTreeUI.java:2159)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.updateExpandedDescendants(BasicTreeUI.java:2014)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI$Handler.treeExpanded(BasicTreeUI.java:4295)
at java.desktop/javax.swing.JTree.fireTreeExpanded(JTree.java:2853)
at java.desktop/javax.swing.JTree.setExpandedState(JTree.java:3766)
at java.desktop/javax.swing.JTree.expandPath(JTree.java:2296)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:99)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:102)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:102)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:102)
at net.sf.openrocket.gui.components.BasicTree.expandChildlessNodes(BasicTree.java:91)
at net.sf.openrocket.gui.components.BasicTree.treeDidChange(BasicTree.java:77)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.updateSize(BasicTreeUI.java:2159)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI.updateExpandedDescendants(BasicTreeUI.java:2014)
at java.desktop/javax.swing.plaf.basic.BasicTreeUI$Handler.treeExpanded(BasicTreeUI.java:4295)
at java.desktop/javax.swing.JTree.fireTreeExpanded(JTree.java:2853)
at java.desktop/javax.swing.JTree.setExpandedState(JTree.java:3716)
at java.desktop/javax.swing.JTree.expandPath(JTree.java:2296)
at net.sf.openrocket.gui.main.componenttree.ComponentTreeModel.fireTreeStructureChanged(ComponentTreeModel.java:124)
at net.sf.openrocket.gui.main.componenttree.ComponentTreeModel.componentChanged(ComponentTreeModel.java:143)
at net.sf.openrocket.rocketcomponent.Rocket.notifyAllListeners(Rocket.java:653)
at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:559)
at net.sf.openrocket.rocketcomponent.Rocket.fireComponentChangeEvent(Rocket.java:568)
at net.sf.openrocket.rocketcomponent.Rocket.thaw(Rocket.java:718)
[... 515 lines cut ...]
158264 378.832 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------
[image]

### To Reproduce

1.Edit EB mass component to center 0 in
2. Drag EB mass component into forward body tube
3.

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

23.09

### What platform are you running on?

Linux

nevinsb, 2023-12-13:
[YIRIS55-N.zip](https://github.com/openrocket/openrocket/files/13658756/YIRIS55-N.zip)

SiboVG, 2023-12-13:
Thanks for your bug report! This is a duplicate of #2418.

closed 2023-12-13 by SiboVG