openrocket/openrocket #2424: [Feature Request] Present Thrust Curve Date data in OR "Show Details" window

jimmiedave, 2023-12-16:
**Is your feature request related to a problem? Please describe.**
While troubleshooting problem with the AT M4500-P ST motor's thrust curve, I wasn't able to find a version or date or serial or certification date in the OR UI that would let me know if what I was looking at was supposed to be the same curve as the one I was looking at on Thrustcurve.

**Describe the solution you'd like**
Where a thrust curve has a certification date, a version number, or a serial-number type record, present that in the "Show Details" window to enable the user to know with certainty that the curve does or does not match a curve outside of OR. In all cases, show the date the thrust curve was added to OpenRocket.

**Describe alternatives you've considered**
Status Quo

**Additional context**
Presenting dates helps increase user confidence that the program is working as expected on the right data, and allows users to assess with some confidence how current the data is. It will also assist in tracking down discrepancies caused by rogue or outdated local file thrust curves.

This might be particularly useful in cases where there are more than one curve available for a motor (sometimes due to local import).

JoePfeiffer, 2023-12-16:
That would be good metadata to add; especially a field saying whether a motor came from the user or from thrustcurve.

JoePfeiffer, 2023-12-16:
Would it also be worth while to somehow show the certification data and add a caveat that a particular curve might not match it? That would give a user a fighting chance at understanding when a curve doesn't match the certification data it isn't our fault.