openrocket/openrocket #2430: Uncaught Exception

mbcullen, 2023-12-30:
---------- Bug report ----------

Please include a description about what actions you were performing when the exception occurred:
(You can edit text directly in this window)

1. Adding a parachute
2.
3.

If possible, please send us the .ork file that caused the bug.

Include your email address (optional; it helps if we can contact you in case we need additional information):

(Do not modify anything below this line.)
---------- Exception stack trace ----------
java.lang.IllegalStateException: Stage not found in copy
at net.sf.openrocket.rocketcomponent.Rocket.copyWithOriginalID(Rocket.java:379)
at net.sf.openrocket.document.OpenRocketDocument.addStateToUndoHistory(OpenRocketDocument.java:531)
at net.sf.openrocket.document.OpenRocketDocument.addUndoPosition(OpenRocketDocument.java:508)
at net.sf.openrocket.gui.adaptors.PresetModel.setSelectedItem(PresetModel.java:70)
at net.sf.openrocket.gui.dialogs.preset.ComponentPresetChooserDialog.applySelectedPreset(ComponentPresetChooserDialog.java:241)
at net.sf.openrocket.gui.dialogs.preset.ComponentPresetChooserDialog$4.actionPerformed(ComponentPresetChooserDialog.java:217)
at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
at java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
at java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at java.desktop/javax.swing.plaf.basic.BasicButtonListener.mouseReleased(BasicButtonListener.java:279)
at java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
at java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
at java.desktop/java.awt.Component.processEvent(Component.java:6391)
at java.desktop/java.awt.Container.processEvent(Container.java:2266)
at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5001)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
at java.desktop/java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4948)
at java.desktop/java.awt.LightweightDispatcher.processMouseEvent(Container.java:4575)
at java.desktop/java.awt.LightweightDispatcher.dispatchEvent(Container.java:4516)
at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2310)
at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2780)
at java.desktop/java.awt.Component.dispatchEvent(Component.java:4833)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:773)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:97)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:746)
at java.desktop/java.awt.EventQueue$5.run(EventQueue.java:744)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:743)
at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:117)
at java.desktop/java.awt.WaitDispatchSupport$2.run(WaitDispatchSupport.java:191)
at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:236)
at java.desktop/java.awt.WaitDispatchSupport$4.run(WaitDispatchSupport.java:234)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:318)
at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:234)
at java.desktop/java.awt.Dialog.show(Dialog.java:1080)
at java.desktop/java.awt.Component.show(Component.java:1728)
at java.desktop/java.awt.Component.setVisible(Component.java:1675)
at java.desktop/java.awt.Window.setVisible(Window.java:1036)
at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1016)
at net.sf.openrocket.gui.configdialog.RocketComponentConfig$7.run(RocketComponentConfig.java:457)
at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:771)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
[... 383 lines cut ...]
1646 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.WaitDispatchSupport.enter(WaitDispatchSupport.java:234)
1647 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Dialog.show(Dialog.java:1080)
1648 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Component.show(Component.java:1728)
1649 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Component.setVisible(Component.java:1675)
1650 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Window.setVisible(Window.java:1036)
1651 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.Dialog.setVisible(Dialog.java:1016)
1652 695.395 STDERR (SwingExceptionHandler.java:49) at net.sf.openrocket.gui.configdialog.RocketComponentConfig$7.run(RocketComponentConfig.java:457)
1653 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
1654 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:771)
1655 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:722)
1656 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:716)
1657 695.395 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
1658 695.395 STDERR (SwingExceptionHandler.java:49) at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
1659 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:741)
1660 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:203)
1661 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
1662 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
1663 695.395 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
1664 695.396 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
1665 695.396 STDERR (SwingExceptionHandler.java:49) at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
1666 695.396 INFO (SwingExceptionHandler.java:61) Exception handler running on EDT, showing dialog
1667 695.396 INFO (SwingExceptionHandler.java:201) Showing Exception dialog
1668 702.730 USER (SwingExceptionHandler.java:220) User requested sending bug report
1669 702.734 DEBUG (JarUtil.java:40) Found jar file using codeSource
---------- End of bug report ----------

JoePfeiffer, 2024-01-07:
Thank you for your report. This is an issue that seems to crop up intermittently -- if you perform the same steps, are you able to get it to happen again?

JoePfeiffer, 2024-01-09:
Superceded by #2435