openrocket/openrocket #2434: Provide saved configurations for plots

neilweinstock, 2024-01-08:
Simulation plots are infinitely configurable.  I suspect it would be useful to allow users to save plot configurations so they can be recalled quickly.

Mostly likely we could add a few pre-configured plots as well.

I would make this global for the app, not per-design.

SiboVG, 2024-01-08:
I guess this is a duplicate of #2162? 

Do you want to add more pre-configured plots than the existing ones?
[image]

neilweinstock, 2024-01-08:
Holy crap I never noticed those preset configurations were there.  Yikes.

In any case yet, this is a duplicate of #2162, I'll close it now.

closed 2024-01-08 by neilweinstock