openrocket/openrocket #2437: [Bug] Disabling sustainer on single stage rocket has no effect

JoePfeiffer, 2024-01-10:
### Describe the bug

When we disable all of the stages in a rocket and simulate it, the simulation should raise an exception. This works correctly for rockets with more than a single stage; a single stage rocket simulates successfully.

It is interesting to note why the unit tests fail to catch this.  DisableStageTest.testSingleStage() checks the exception that is thrown to make sure it's a MotorIgnitionException, but it doesn't actually check to see that the exception is thrown at all.

### To Reproduce

1. Open "Simple Model Rocket" example
2. Disable sustainer. Rocket disappears from design window
3. Run first simulation. Simulation runs successfullly

Compare with the result of 
1. Open "Two Stage Rocket" example
2. Disable both stages. Rocket disappears from design window
3. Run first simulation. Simulation throws an exception.

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

23.09, unstable. Haven't checked 22.02.

### What platform are you running on?

Linux

JoePfeiffer, 2024-01-11:
I'm not entirely sure it's doing the right thing on multistage, either after a little more looking...

SiboVG, 2024-12-25:
> I'm not entirely sure it's doing the right thing on multistage, either after a little more looking...

What do you mean?