openrocket/openrocket #2438: [Bug] exception in RockSim export

JoePfeiffer, 2024-01-13:
### Describe the bug

User "CP Fleck" <cfleck@mi.rr.com> reports a null pointer exception exporting to RockSim
I've confirmed the exception still occurs in current Unstable.

### To Reproduce

1. 
2. 
3. 
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[SLS Block I.V2.zip](https://github.com/openrocket/openrocket/files/13929534/SLS.Block.I.V2.zip)

### OpenRocket version

23.09

### What platform are you running on?

Windows

### Additional context

java.lang.NullPointerException: Cannot invoke "String.length()"
because "str" is null at
java.base/java.io.Writer.write(Writer.java:249) at
net.sf.openrocket.file.rocksim.export.RockSimSaver.save(RockSimSaver.java:67)
at
net.sf.openrocket.file.GeneralRocketSaver.saveInternal(GeneralRocketSaver.java:245)
at
net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:165)
at
net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:113)
at
net.sf.openrocket.file.GeneralRocketSaver.save(GeneralRocketSaver.java:74)
at
net.sf.openrocket.gui.main.BasicFrame.saveRockSimFile(BasicFrame.java:1630)
at
net.sf.openrocket.gui.main.BasicFrame.saveAsRockSim(BasicFrame.java:1618)
at
net.sf.openrocket.gui.main.BasicFrame.exportRockSimAction(BasicFrame.java:1577)
at
net.sf.openrocket.gui.main.BasicFrame$7.actionPerformed(BasicFrame.java:439)
at
java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
at
java.desktop/javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2313)
at
java.desktop/javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:405)
at
java.desktop/javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:262)
at
java.desktop/javax.swing.AbstractButton.doClick(AbstractButton.java:374)
at
java.desktop/javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:1028)
at
java.desktop/javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:1072)
at
java.desktop/java.awt.Component.processMouseEvent(Component.java:6626)
at
java.desktop/javax.swing.JComponent.processMouseEvent(JComponent.java:3389)
at java.desktop/java.awt.Component.processEvent(Component.java:6391)
[... 715 lines cut ...]
 

	 198488 3950.417 STDERR (SwingExceptionHandler.java:49) at
java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
 

	 198489 3950.417 STDERR (SwingExceptionHandler.java:49) at
java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
 

	 198490 3950.417 STDERR (SwingExceptionHandler.java:49) at
java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
 

	 198491 3950.417 INFO (SwingExceptionHandler.java:61) Exception
handler running on EDT, showing dialog  

	 198492 3950.417 INFO (SwingExceptionHandler.java:201) Showing
Exception dialog  

	 198493 3960.666 USER (SwingExceptionHandler.java:220) User requested
sending bug report  

	 198494 3960.674 DEBUG (JarUtil.java:40) Found jar file using
codeSource

SiboVG, 2024-01-18:
Interesting, the bug is caused by the boosters in the design. Does RockSim even support boosters?

SiboVG, 2024-01-18:
I guess a booster in RockSim is a pod that can be ejected during simulation?
[image]

PR #2447 (2024-01-28): [#2438, #2377, #2435] Support RockSim booster importing/exporting, pods/booster splitting & fix a bunch of bugs
Okay, what started out as a small PR to fix #2438 quickly turned into much more... I'll take you through my rollercoaster.

The issue with #2438 was that exporting booster designs to RockSim would throw an error because booster exporting simply wasn't implemented. So, I started implementing it.

Then I noticed that OR only exported one instance of the pods/boosters. Also, the rotation of children components of the pod set was incorrect. That needed fixing. To do that, I needed to spl [... cut]
closed 2024-01-28 by JoePfeiffer