openrocket/openrocket #2443: [Bug] Acceleration is zero in SimulationListener's SimulationStatus

Graicc, 2024-01-20:
### Describe the bug

The acceleration in `SimulationStatus.getRocketAcceleration()` is `(0,0,0)` until apogee for all simulation listeners. On unstable (tested on `b9084`), `postAccelerationCalculation` contains the acceleration data, but the status still contains zero.  On 23.09, the acceleration data in `postAccelerationCalculation` is null.

### To Reproduce

```java
public class MyListener extends AbstractSimulationListener {
	@Override
	public void postStep(SimulationStatus status) throws SimulationException {
		var accel = status.getRocketAcceleration(); // is (0,0,0)
	}
	
	@Override
	public AccelerationData postAccelerationCalculation(SimulationStatus status, AccelerationData acceleration) throws SimulationException {
		var goodAccel = acceleration.getLinearAccelerationWC(); // Correct
		var accel = status.getRocketAcceleration(); // is (0,0,0)
	}
}
```

### Screenshots / .ork file

[image]
Only the z component is shown in the screenshot, but all fields were 0.

### OpenRocket version

unstable

### What platform are you running on?

Windows

### Additional context

Migrated from conversation in the OpenRocket Discord here: https://discord.com/channels/1073297014814691328/1078118846818439199/1197974403049336892

The last message, from @JoePfeiffer:
> As you've discovered, the RK4 stepper doesn't set the acceleration in the simulation status; that doesn't end up getting used until the landing stepper starts up. I'm not sure at the moment whether the bug is not putting the acceleration data in the status or if it's having the place in the status to put it, but it's clearly one or the other. The store actually seems like the better place to have it anyway, so go ahead and keep using that.

SiboVG, 2024-01-20:
Thanks for filing the issue!

JoePfeiffer, 2024-01-20:
Storage of simulation variables is kind of a mess right now. SimulationStatus mainly contains what I think of as status variables, like the event queue, whether we've cleared the launch rod, and stuff like that. It also contains things like the current rocket position, and thinks it contains the current acceleration. For the RK4 stepper, we both have an RK4SimulationStatus that extends SimulationStatus and adds a few things that seem to me like they belong in SimulationStatus (like max Z acceleration encountered), and adds a private static DataStore class that contains the data necessary for the stepper (like the acceleration, thrust force, and more). Meanwhile, the "official" place all the data is stored after a simulation step is in a FlightDataBranch. I cleaned up (and fixed) a little of this last year (see #2352), but am hoping to do more after NARCON.

The Euler stepper actually does put acceleration data in the SimulationStatus, something I should have fixed back when I refactored it last year, but didn't.

This, along with improving simulation step size calculation, is the main stuff I want to work on right now.

SiboVG, 2024-07-12:
@JoePfeiffer I see you self-assigned this issue; was this issue fixed by one of your PRs?

JoePfeiffer, 2024-07-12:
No, working on it now.

PR #2514 (2024-07-18): Reduce redundant acceleration data storage
The DataStore, AccelerationData, and SimulationStatus had redundant storage of identical acceleration data. This made the code more difficult to read and has a potential for errors in the future if the data gets out of sync. This PR removes this redundancy. Issue #2443 was originally filed because some of this data was only in the SimulationStatus when in an Euler stepper, not the RK4 Stepper.  It is now never in SimulationStatus (it is in DataStore.accelerationData at all times).

Fixes #2443
closed 2024-07-18 by SiboVG