openrocket/openrocket #2454: [Bug] Some component default wall thicknesses are 0

JoePfeiffer, 2024-02-02:
### Describe the bug

Default wall thickness of coupler and engine block, and (if a shoulder is added to a nose cone) shoulder diameter and thickness for nose cone are all 0.

### To Reproduce

1. Create a nose cone
2. Extend the shoulder 1 cm
3. Diameter and thickness are 0. It would be better if they were (nose cone diameter) - 2*default body tube diameter
4. Add a body tube
5. Add a coupler or engine block to the body tube. Their wall thicknesses are 0. Note this is particularly misleading for the engine block, since the length is non-zero and in side view it looks like there's something there.
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

22.02, 23.09, unstable

### What platform are you running on?

Linux

### Additional context

This is on the border between a bug report and a feature request. Either way, should be fixed.

SiboVG, 2024-12-26:
> 3\. nose cone diameter) - 2*(default body tube diameter), and thickness was (default body tube diameter)

Shouldn't this be `shoulder diameter = (nose cone diameter) - (nose cone thickness)` and `thickness = (nose cone thickness) / 2`? That way, the shoulder starts halfway the cone wall and ends at the inner wall. I'm not sure if that's the most common configuration in real life nose cones.

JoePfeiffer, 2024-12-26:
Yikes, what was I thinking? What I wrote up there is obvious nonsense. 

Looking at a low power nose cone from my parts box, there isn't any obvious internal "step", the thickness of the shoulder is the thickness of the nose cone minus the thickness of the matching body tube. So a PNC-50KA nose cone has a base diameter of 2.48cm and a wall thickness of .157cm; it mates with a BT-50A body tube with a .033cm wall thickness so the shoulder thickness should be .157-.033=.124mm. Unfortunately the shoulder thickness OR calculates for it is 0.