openrocket/openrocket #2455: [Bug] <GUI glitching out>

ServerDummy, 2024-02-05:
### Describe the bug

The GUI is glitching out, I am on windows 11, uninstalled and reinstalled latest java and graphics drivers, I have had this issue for almost a year now.

https://github.com/openrocket/openrocket/assets/63013446/3b064992-d486-4112-9e54-5845ca0265dc

### To Reproduce

1. open rocket
2. click menu
3. breaks?
...

### Screenshots / .ork file

#### Screenshot(s):
*[image]*

#### .ork file:
*(drag-and-drop the file here as a .zip file)*

### OpenRocket version

23.09

### What platform are you running on?

Windows

jet689, 2024-02-18:
yeah I get this to interestingly I get the problem on my pc but not my laptop.

jet689, 2024-02-18:
Actualy yeah I do have a Nvidia GPU. I did use the nvidia driver installed from their driver installer, I will check if there is an update but maybe reverting to the windows recomended might do something idk.

jet689, 2024-02-18:
nope tried all drives nividia game ready studio and windows rec nothing fixed it. looks like the same bug as [https://github.com/openrocket/openrocket/issues/1667](url)

ServerDummy, 2024-02-18:
If you have integrated graphics on your cpu with your desktop maybe try that. I can give it a try in a few days

jet689, 2024-02-18:
nope unforunatly.

jet689, 2024-02-19:
okay so it looks like if you find OpenRocket.exe (it will probalbly be under Program files and openrocket) then you right click it choose properties then the compatability tab and select run this program as administrator it works and it will automatically ask you for administrator permission before running.

SiboVG, 2024-08-22:
Duplicate of #1667.

closed 2024-08-22 by SiboVG