openrocket/openrocket #2463: [Bug] Edit>Scale (downscale at least) appears to over-affect NC shoulder

jimmiedave, 2024-03-05:
### Describe the bug

Scaled a rocket design to 75% (selected components only) and excluded only MMT and CRs. (Nose cone was selected)

Everything else scaled as expected, but the nose cone shoulder was scaled down in diameter past 75% of the original. Calculated shoulder diameter should probably be 57.15mm diameter, but shoulder is scaled to 44.4mm diameter. Shoulder length scales as expected.

Upscale does not appear to cause a problem.

### To Reproduce

1. Load a design
2. Select all the components
3. Deselect MMT and its children (might not be necessary)
4. Edit>Scale
5. Select "Only selected component(s)"
6. Click "Scale"
7. Observe nose cone shoulder.
...

### Screenshots / .ork file

#### Screenshot(s):
*(drag-and-drop the screenshot(s) here)*
[image]

[image]

#### .ork file:
*(drag-and-drop the file here as a .zip file)*
[ArgentFG54v01.ork.zip](https://github.com/openrocket/openrocket/files/14500761/ArgentFG54v01.ork.zip)

### OpenRocket version

23.09

### What platform are you running on?

macOS

AhanuDewhirst, 2024-07-21:
Hello. I'd like to work on this bug.

SiboVG, 2024-07-21:
> Hello. I'd like to work on this bug.

Great, thanks!

PR #2563 (2024-09-26): [#2463] Fix shoulder radius scaling issue when smaller than 1 + fin tab length/offset scaling
This PR fixes #2463. Additionally, it also fixes fin tab length/offset scaling.
closed 2024-09-26 by SiboVG