openrocket/openrocket #2474: .01" tall fins break the CP calculation

neilweinstock, 2024-03-30:
If you put fins on a rocket that are .01" tall, CP becomes N/A.  If you make them 0.1" tall, CP calculates normal (assume there are other fins, these are not the only ones).

Here's the Simple Model Rocket example with a set of these fins.

[image]
[short_fins.ork.zip](https://github.com/openrocket/openrocket/files/14809281/short_fins.ork.zip)

JoePfeiffer, 2024-03-30:
Is this still the case in current unstable?

neilweinstock, 2024-03-30:
Not sure, I'm still not yet on the unstable.  Can you try it?

JoePfeiffer, 2024-03-30:
Checking... yes, it's still there. Darn.

SiboVG, 2024-12-23:
Some further context:

- The issue is the `Roll forcing coefficient` which becomes `NaN` because:
- `macSpan` in `calculateNonaxialForces` `forces.setCrollForce((macSpan + r) * cna1 * (1 + tau) * cantAngle / conditions.getRefLength());` in `FinSetCalc.java` is `NaN` because:
- `area` in `calculateFinGeometry()` `macSpan /= area;` in `FinSetCalc.java` is 0 because:
- `length` is always 0  in the `for (int i = 0; i < DIVISIONS; i++) {
			double length = chordTrail[i] - chordLead[i];` loop because:
- `chordTrail` and `chordLead` are both filled with zero's because the height of the fin is so small

SiboVG, 2024-12-23:
Another note: if the fin height is set to 0 and the fin is canted, the roll forcing coefficient is NaN, and the Friction CD is infinity.

PR #2633 (2025-01-09): [#2474] Add extra sanity checks for small or zero-area fins in aerodynamics calculations
This PR fixes #2474 by adding extra sanity checks in `FinSetCalc` for when the fin area/MAC span is zero or very small. See also [this comment](https://github.com/openrocket/openrocket/issues/2474#issuecomment-2558826495).
closed 2025-01-09 by SiboVG