openrocket/openrocket #2480: Parts library error: plastic nose cone shown as "filled"

neilweinstock, 2024-05-09:
Estes part #72413 (9.75" x 2.5" plastic nose cone) shows as "filled". I don't know what the correct wall thickness should be but I'm pretty certain it should be hollow.
[image]

SiboVG, 2024-05-09:
Should we move component database issue to [here](https://github.com/openrocket/openrocket-database/issues) from now on?

JoePfeiffer, 2024-05-09:
> Should we move component database issue to [here](https://github.com/openrocket/openrocket-database/issues) from now on?

I'd say so. Note the problem appears to be that the wall thickness is .125 with no units, so it's being limited to the radius of the nose cone and marked as "filled" (even though it's not marked as filled in the database). 72653 has the same problem (but not the other components in that file).

closed 2024-12-18 by JoePfeiffer