openrocket/openrocket #2485: [Feature Request] Hide components

SiboVG, 2024-05-26:
**Is your feature request related to a problem? Please describe.**
Sometimes, designs can get visually cluttered by components. I'd like a way to visually hide components. The hidden components would still account for simulations etc., but would just not show in the rocket design view.

**Describe the solution you'd like**
An option in the Edit menu and context menu to Hide the selected component. Once a component is hidden, the Edit menu item would change from "Hide selected component" to "Unhide selected component". There should also be an option to Unhide all components.

Components that are hidden should have greyed out text in the component tree and a hidden icon:
[image]

AhanuDewhirst, 2024-07-14:
Hello. I'd like to work on this feature.

SiboVG, 2024-07-15:
> Hello. I'd like to work on this feature.

Hello, that's great! Be sure to check out our developer's guide to get you started: [https://openrocket.readthedocs.io/en/latest/dev_guide/development_overview.html](https://openrocket.readthedocs.io/en/latest/dev_guide/development_overview.html).

JoePfeiffer, 2024-07-15:
Welcome aboard! As @SiboVG says, check out the developer's guide. Also, please feel free to ask questions about anything that's unclear. We'll be more than happy to help you get started.

AhanuDewhirst, 2024-07-16:
Thank you both for pointing me in the right direction. I've familiarised myself with the developer's guide :smile:

Addressing this feature, I've taken the following approach:

1. Component visibility can be toggled through the Edit and Context menus, or using the command `CTRL + ,`.
2. A hidden state is visually indicated by greyed out text and a hidden icon.
3. The visibility of multiple components can be toggled at once.
4. Toggling the visibility of a rocket or stage component affects the visibility of its child components.

Here's what this looks like using a simple model rocket:

[image]

[image]

Let me know what you think!

SiboVG, 2024-07-16:
Already looks pretty good @AhanuDewhirst, great job! The best thing you can do now is to create a Pull Request so we can check your code and test the functionality.

> 1. or using the command `CTRL + ,`

I wouldn't do this, since this is the shortcut to open the preference dialog.

AhanuDewhirst, 2024-07-16:
Will do!

> I wouldn't do this, since this is the shortcut to open the preference dialog.

Noted. I'll remove the shortcut for now.

PR #2511 (2024-07-19): [#2485] Hide components
This PR fixes #2485 by allowing component visibility to be changed in the component tree via the Edit and Context menus.

[image]

[image]
closed 2024-07-19 by JoePfeiffer