openrocket/openrocket #2490: [Enhancement] Use Crowdin to host the translations

SiboVG, 2024-06-01:
OpenRocket's translation management is not good. Many translations are incomplete and out-of-date and it's hard to know which parts of the translation needs updating.

I propose we move everything to [Crowdin](https://crowdin.com/), which is easy to use (so volunteers can easily contribute instead of needing to know how GitHub works), shows you how up-to-date translations are, and it's free for open-source projects.

andrii-bodnar, 2024-06-20:
Hello, I just landed here by accident and was impressed, I've never seen software like this before 😄

This is a great idea, Crowdin handles `properties` files very well, so the integration should be easy.

Please let me know if you have any questions about the integration process!

SiboVG, 2024-06-20:
> Hello, I just landed here by accident and was impressed, I've never seen software like this before 😄
> 
> This is a great idea, Crowdin handles `properties` files very well, so the integration should be easy.
> 
> Please let me know if you have any questions about the integration process!

Hi @andrii-bodnar, honor to have you here! OpenRocket is indeed a fun piece of software 🙂

Great to hear that `properties` files are well-supported. I will keep you in mind in case we have troubles. Thanks for responding!

SiboVG, 2025-07-15:
You can find the Crowdin project here: https://crowdin.com/project/openrocket